renode/renode #261: Button input via GPIO0 is not working on Nordic NRF52840

RogeHong, 2021-11-05:
Hi,

I add a button input to GPIO0 pin 11 as Nordic NRF52840 DK board below. But when execute command, the running simulator cannot get the input signal. 

The same source code with GPIO PIN definition update is working with STM32F4 discovery board in renode so the source should be OK. 

Please confirm whether GPIIO input is working with NRF52840 soc input or  not. If yes, what is the correct procedure to capture the external GPIO input in Renode for NRF52840.

Thanks.
Regards,
Roger 

using "platforms/cpus/nrf52840.repl"

UserButton: Miscellaneous.Button @ gpio0 11
    -> gpio0@11

sysbus.gpio0.UserButton Press

mateusz-holenko, 2021-11-09:
Hi @RogeHong, thanks for reporting the issue.

We are currently working on some fixes to the GPIO controller for this platform. I'll let you know when they are merged upstream.

RogeHong, 2021-11-11:
Hi, Mateusz:

Thanks for your information. I will keep in tune for that. Thanks.

Regards,
Roger Hong

mateusz-holenko, 2021-12-15:
@RogeHong https://github.com/renode/renode/commit/c01040f626d5b3a37d6d3abc005841e239fb6818 introduced fixes to the NRF GPIO controller model - could you check if the current version works for you?

RogeHong, 2021-12-21:
> @RogeHong <https://github.com/RogeHong> c01040f

Hi, Mateusz:

Sorry for my late reply. I have updated to latest renode
version fb16507c-202112170224 but it still doesn't work. Here you are my
information. Let me know any further information you need to debug this
issue.

Thanks.

Regards,
Roger Hong

1. NRF52840 description file:
using "platforms/cpus/nrf52840.repl"

UserButton: Miscellaneous.Button @ gpio0
    -> ***@***.***

2. The hardware architecture
sysbus (SystemBus)
  │
  ├── clock (NRF52840_CLOCK)
  │       <0x40000000, 0x40000FFF>
  │
  ├── cpu (CortexM)
  │       Slot: 0
  │
  ├── ecb (NRF52840_ECB)
  │       <0x4000E000, 0x4000EFFF>
  │
  ├── flash (MappedMemory)
  │       <0x00000000, 0x000FFFFF>
  │
  ├── gpio0 (NRF52840_GPIO)
  │   │   <0x50000500, 0x500007FF>
  │   │
  │   └── UserButton (Button)

3. The command usage:

(machine-0) sysbus.gpio0.UserButton Press
(machine-0) sysbus.gpio0.UserButton Release

4. Sample code without any changes from Zephyr :
zephyr/samples/basic/button

mateusz-holenko, 2021-12-21:
I forgot to add an important information that might be vital here, sorry.

In the NRF52840 DK platform buttons are active low - they are connected to ground.
That's why in Renode they need to be marked as `inverted`: https://github.com/renode/renode/commit/c01040f626d5b3a37d6d3abc005841e239fb6818#diff-e809ddeec5b84e358837cdc7f0c4caf4529657725b73936dc24e9cb9506a8595R60.

With this change we are able to run the Zephyr button sample successfully in Renode.

mateusz-holenko, 2022-04-05:
Closing this issue for now, but if you have any other problems feel free to open a new one.

closed 2022-04-05 by mateusz-holenko