renode/renode #262: Freature Request: ESP32 and RP2040 support

DatanoiseTV, 2021-11-06:
I would love to see some ESP32 and RP2xxx support. Is this planned?

jjpark78, 2021-11-07:
we need this feature too

Darneas, 2022-04-16:
Interested as well.

mithro, 2022-04-16:
Some initial support for the Xtensa architecture used in the ESP32 has been started -- see blog post at https://antmicro.com/blog/2022/01/xtensa-isa-in-renode-for-sof-project/

chirambaht, 2023-09-28:
Hi all, have there been any updates to this? I have struggled to get an ESP32 board up and running.

lptr, 2024-03-28:
Is there any progress on ESP32-S3 support?

bjk2k, 2024-05-09:
I am also very interested in ESP32-S3 support.

JaimeAlbq, 2024-07-08:
It would be awesome to have Renode supporting all Espressif's MCUs, since both have the goal to be implemented in IoT projects.

> Some initial support for the Xtensa architecture used in the ESP32 has been started -- see blog post at https://antmicro.com/blog/2022/01/xtensa-isa-in-renode-for-sof-project/

@mithro, I can see some scripts and configurations for xtensa, but it seems specific for Zephyr project only, which I am not using on my project for now yet, but I still want to simulate using the ESP-IDF, and if it possible, in PlatformIO, that would be my perfect combo!

But anyway, I think for now we can try to use QEMU and whenever Renode gets enough support to ESP32, I will be very happy to switch.