renode/renode #265: Got unhandled exception when running test script on windows.

jjpark78, 2021-11-12:
used script 
```
using sysbus

mach create
machine LoadPlatformDescription @platforms/cpus/stm32f103.repl
machine LoadPlatformDescriptionFromString "button: Miscellaneous.Button @ gpioPortC 13 { IRQ -> gpioPortC@13 }"

$bin?=@https://dl.antmicro.com/projects/renode/zephyr-stm32f103-button.elf-s_276760-1bf32c99bbb3c01d81e13ca68118eaf08b2a815f

showAnalyzer usart1

macro reset
"""
    sysbus LoadELF $bin
"""

runMacro $reset
```

and everytime i clicked renode terminal window, it throw a following exception. 

메서드를 찾을 수 없습니다. 'Void Xwt.Clipboard.SetPrimaryText(System.String)'
   위치: TermSharp.Terminal.OnCanvasButtonReleased(Object sender, ButtonEventArgs e)
   위치: Xwt.Widget.OnButtonReleased(ButtonEventArgs args)
   위치: Xwt.WPFBackend.WidgetBackend.<>c__DisplayClass84_0.<WidgetMouseUpHandler>b__0()
   위치: Xwt.Toolkit.Invoke(Action a)

PiotrZierhoffer, 2021-11-23:
Hi @jjpark78 

This is quite a peculiar error. How do you obtain Renode? Is there any way some DLLs could have been taken from different Renode versions?

Do you observe the same behavior on our nightly builds? https://dl.antmicro.com/projects/renode/builds/renode-latest.msi

mateusz-holenko, 2021-11-23:
I checked the current release (1.12), the nightly build and current master (built from sources) and couldn't reproduce the problem in any case.

lcavalli, 2021-11-26:
I'm running latest windows nightly build taken from https://dl.antmicro.com/projects/renode/builds/ (renode-latest.msi 2021-11-25 11:28   18M) and I'm experiencing the very same issue.

lcavalli, 2021-11-26:
Suspecting the issue was due to commit e43075b388117ccd87a5c116a23bc656ed185672 I installed a version older than Sep 28, 2021 and it worked as expected. Then installed later versions, but none of them presented the issue. I finally re-installed renode-latest.msi 2021-11-25 11:28 18M and the issue is no more present. It seems fixed. Before installing each version I removed the previous one (no upgrade).

jjpark78, 2021-12-08:
thanks. as @lcavalli mentioned, this issue does not seems to be present in latest release. 

thanks.

closed 2021-12-29 by mateusz-holenko