renode/renode #266: Test with renode and code coverage report

jjpark78, 2021-11-18:
i just curious about when testing with renode using Robot Framework, renode can measure code coverage or not

when develop web apps, e2e test can generate some of coverage report and mocking environment is not that hard

but in my opinion, unit test in embedded is too expensive, cause of need to mocking everything

so, i thinks renode can do this, it will be really great !!!

PiotrZierhoffer, 2021-11-23:
Hi @jjpark78 

Tracking execution is something that can be easily done with Renode, but with the abundance of coverage tools out there it's difficult to select just one.

You can call `cpu SetHookAtBlockBegin` and `cpu SetHookAtBlockEnd` in your script, providing it with a Python script that will log specific information. In these scripts you have access to the following variables:

- `pc` - current program counter
- `size` - number of instructions
- `cpu` - giving you access to most of the CPU methods

For example

```
cpu SetHookAtBlockBegin "print(str(pc) +' ' +str(size))"
```

If you are interested in any particular coverage reporting tool, please drop us an email.

mateusz-holenko, 2021-12-15:
FYI @jjpark78: In https://github.com/renode/renode/commit/2a79488d8126dfa9edd91a0666720ddb8938fe0f we added the execution tracer module that can dump executed PCs / opcodes to a separate file.

jjpark78, 2021-12-18:
> FYI @jjpark78: In [2a79488](https://github.com/renode/renode/commit/2a79488d8126dfa9edd91a0666720ddb8938fe0f) we added the execution tracer module that can dump executed PCs / opcodes to a separate file.

that a good update. 

how can i use this from robot test framework ?

mateusz-holenko, 2021-12-28:
At the moment the module supports dumping a trace to a file, so you need to search the file content as in: https://github.com/renode/renode/blob/master/tests/unit-tests/execution-tracing.robot#L27.

PiotrZierhoffer, 2022-02-01:
Closing the issue for now, feel free to reopen if you need further input.

closed 2022-02-01 by PiotrZierhoffer

mkschreder, 2022-10-17:
@PiotrZierhoffer how can we get an lcov coverage report from the trace?