renode/renode #267: Nothing working on Fedora 35

damien-lemoal, 2021-11-19:
Was Renode checked with Fedora 35 ? Nothing is working. All scripts throw:

[ERROR] FATAL UNHANDLED EXCEPTION: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.DllNotFoundException: dl assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Antmicro.Renode.Utilities.SharedLibraries.dlerror()
...

damien-lemoal, 2021-11-19:
Note: the portable tar.gz code does not even start:

/lib64/libpthread.so.0: version `GLIBC_PRIVATE' not found (required by ./renode)

mateusz-holenko, 2021-11-23:
Thanks for reporting the issue, @damien-lemoal !

We'll take a look at this.

damien-lemoal, 2021-11-23:
Note: I did install mono-complete from the Mono project. I did not use the default Fedora packages. I tried the portable version too, but that one has a problem with libraries (not found on startup). I tried the pre-compiled rpm and compiling from source. Both give the same result: error "dl assembly...". Compilation does not generate any error but has some warnings.

I am going to try on Fedora 34 today, just to check if anything is different.

damien-lemoal, 2021-11-23:
I just tried the precompiled rpm on Fedora 34 (everything uptodate), after installing mono from the mono project.
Everything is working fine. Compiling from source is also OK and works fine.

So it looks like there is something wrong with Fedora 35. No clue what though.

mateusz-holenko, 2021-12-02:
Might be related to what is visible in Ubuntu 21.10 - please take a look at a temporary workaround I proposed in https://github.com/renode/renode/issues/273#issuecomment-984414345.

damien-lemoal, 2021-12-02:
Indeed, no libdl.so symbolic link exists.

ln -s /usr/lib64/libdl.so.2 /usr/lib64/libdl.so

Fixed the issue.

Thanks for the hint. Closing this issue.

closed 2021-12-02 by damien-lemoal