renode/renode #273: Fatal error (Got unhandled exception) on Kubuntu 21.10

haliskilic, 2021-11-28:
Hi, 

I'm trying to start the demo project. When I execute the demo project command, I see this fatal error. I shared all information.

Step 1 -> start renode with console
`~$ renode`
Step 2 -> start stm32f4_discovery.resc with monitor
`s @scripts/single-node/stm32f4_discovery.resc`
Result -> fatal error occurs !

**Fatal error**
```
Exception has been thrown by the target of an invocation.
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x0001a] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.RuntimeConstructorInfo.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00086] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.RuntimeConstructorInfo.Invoke (System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.ConstructorInfo.Invoke (System.Object[] parameters) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.CreateAndHandleError (System.Reflection.ConstructorInfo constructor, System.Collections.Generic.IEnumerable`1[T] attributes, System.String friendlyName, Antmicro.Renode.PlatformDescription.Syntax.IWithPosition responsibleSyntaxElement) [0x0004a] in <ce00b40eec8b4849b4b56fba26542e5b>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.CreateFromEntry (Antmicro.Renode.PlatformDescription.Syntax.Entry entry) [0x00034] in <ce00b40eec8b4849b4b56fba26542e5b>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessInner (System.String file, System.String source) [0x00061] in <ce00b40eec8b4849b4b56fba26542e5b>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessFile (System.String path) [0x00031] in <ce00b40eec8b4849b4b56fba26542e5b>:0 
  at Antmicro.Renode.PlatformDescription.UserInterface.PlatformDescriptionMachineExtensions.LoadPlatformDescription (Antmicro.Renode.Core.Machine machine, System.String platformDescriptionFile) [0x00006] in <ce00b40eec8b4849b4b56fba26542e5b>:0 
  at (wrapper dynamic-method) System.Object.CallSite.Target(System.Runtime.CompilerServices.Closure,System.Runtime.CompilerServices.CallSite,object,object,object)
  at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2] (System.Runtime.CompilerServices.CallSite site, T0 arg0, T1 arg1, T2 arg2) [0x0011e] in <5b415632df1f4365ae2242b1a257bb5b>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction (System.Runtime.CompilerServices.CallSite& callsite, System.Type binderType, System.Int32 knownType, Dynamitey.Internal.Optimization.InvokeHelper+LazyBinder binder, Dynamitey.InvokeMemberName name, System.Boolean staticContext, System.Type context, System.String[] argNames, System.Object target, System.Object[] args) [0x000df] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite (System.Object target, Dynamitey.InvokeMemberName name, System.Object[] args, System.String[] tArgNames, System.Type tContext, System.Boolean tStaticContext, System.Runtime.CompilerServices.CallSite& callSite) [0x0004f] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Dynamic.InvokeMemberAction (System.Object target, Dynamitey.String_OR_InvokeMemberName name, System.Object[] args) [0x00018] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext (Dynamitey.InvokeContext context, System.Reflection.MethodInfo method, System.Object[] parameters) [0x00023] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeExtensionMethod (System.String name, System.Reflection.MethodInfo method, System.Collections.Generic.List`1[T] parameters) [0x00034] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction (System.Type type, System.String name, System.Collections.Generic.IEnumerable`1[T] p) [0x002d8] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction (System.Type device, System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00023] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName (System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00035] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x00165] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001df] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x0019d] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.TryExecuteScript (System.String filename, AntShell.Commands.ICommandInteraction writer) [0x0016e] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.<InitCommands>b__26_6 (System.String x) [0x00000] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Commands.IncludeFileCommand.Run (AntShell.Commands.ICommandInteraction writer, Antmicro.Renode.UserInterface.Tokenizer.PathToken path) [0x000a7] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Commands.StartCommand.Run (AntShell.Commands.ICommandInteraction writer, Antmicro.Renode.UserInterface.Tokenizer.PathToken path) [0x00000] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at (wrapper managed-to-native) System.Reflection.RuntimeMethodInfo.InternalInvoke(System.Reflection.RuntimeMethodInfo,object,object[],System.Exception&)
  at System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0006a] in <533173d24dae460899d2b10975534bb0>:0 
--- End of stack trace from previous location where exception was thrown ---

  at Antmicro.Renode.UserInterface.Monitor.RunCommand (AntShell.Commands.ICommandInteraction writer, Antmicro.Renode.UserInterface.Commands.Command command, System.Collections.Generic.IList`1[T] parameters) [0x00421] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x001f3] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001df] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x0019d] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at Antmicro.Renode.UserInterface.Monitor.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ci) [0x00000] in <02de5abe86224fdd84249c77fa7a4f44>:0 
  at AntShell.Shell.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ic) [0x000bd] in <6f44d77a9e154853ba700125a787191b>:0 
  at AntShell.CommandLine.HandleControlSequence (AntShell.Helpers.ControlSequence seq) [0x00a45] in <6f44d77a9e154853ba700125a787191b>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.HandleInput (System.Object input) [0x0002e] in <6f44d77a9e154853ba700125a787191b>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.Run (System.Boolean stopOnError) [0x00017] in <6f44d77a9e154853ba700125a787191b>:0 
  at AntShell.Shell.Start (System.Boolean stopOnError) [0x00102] in <6f44d77a9e154853ba700125a787191b>:0 
  at Antmicro.Renode.UI.CommandLineInterface+<>c__DisplayClass0_1.<Run>b__3 (System.Object x) [0x00000] in <ae212e6ac6f74d51bff517caacd44aac>:0 
[... cut ...]
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <533173d24dae460899d2b10975534bb0>:0
```
**Renode Information**
`Renode, version 1.12.0.37823 (44d6786a-202104022100)`

**System Information**
```
Operating System: Kubuntu 21.10
KDE Plasma Version: 5.22.5
KDE Frameworks Version: 5.86.0
Qt Version: 5.15.2
Kernel Version: 5.13.0-22-generic (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-2600 CPU @ 3.40GHz
Memory: 7.7 GiB of RAM
Graphics Processor: AMD TURKS
```
[image]

PiotrZierhoffer, 2021-11-30:
Thanks for the report!

This seems to be related to https://github.com/renode/renode/issues/267

We can reproduce the Fedora issue and we're looking to mitigate it.

Meanwhile, can you please test one of the latest builds from here? https://builds.renode.io/

haliskilic, 2021-11-30:
> Thanks for the report!
> 
> This seems to be related to #267
> 
> We can reproduce the Fedora issue and we're looking to mitigate it.
> 
> Meanwhile, can you please test one of the latest builds from here? https://builds.renode.io/

You are welcome.

I tested the new version. 

`Renode, version 1.12.0.4334 (fabd4209-202111300224)`

The problem is continuing.

Output: 

```
Exception has been thrown by the target of an invocation.
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x0001a] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.RuntimeConstructorInfo.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00086] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.RuntimeConstructorInfo.Invoke (System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.ConstructorInfo.Invoke (System.Object[] parameters) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.CreateAndHandleError (System.Reflection.ConstructorInfo constructor, System.Collections.Generic.IEnumerable`1[T] attributes, System.String friendlyName, Antmicro.Renode.PlatformDescription.Syntax.IWithPosition responsibleSyntaxElement) [0x0004a] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.CreateFromEntry (Antmicro.Renode.PlatformDescription.Syntax.Entry entry) [0x00034] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessInner (System.String file, System.String source) [0x00061] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessFile (System.String path) [0x00031] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.UserInterface.PlatformDescriptionMachineExtensions.LoadPlatformDescription (Antmicro.Renode.Core.Machine machine, System.String platformDescriptionFile) [0x00006] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at (wrapper dynamic-method) System.Object.CallSite.Target(System.Runtime.CompilerServices.Closure,System.Runtime.CompilerServices.CallSite,object,object,object)
  at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2] (System.Runtime.CompilerServices.CallSite site, T0 arg0, T1 arg1, T2 arg2) [0x0011e] in <5b415632df1f4365ae2242b1a257bb5b>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction (System.Runtime.CompilerServices.CallSite& callsite, System.Type binderType, System.Int32 knownType, Dynamitey.Internal.Optimization.InvokeHelper+LazyBinder binder, Dynamitey.InvokeMemberName name, System.Boolean staticContext, System.Type context, System.String[] argNames, System.Object target, System.Object[] args) [0x000df] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite (System.Object target, Dynamitey.InvokeMemberName name, System.Object[] args, System.String[] tArgNames, System.Type tContext, System.Boolean tStaticContext, System.Runtime.CompilerServices.CallSite& callSite) [0x0004f] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Dynamic.InvokeMemberAction (System.Object target, Dynamitey.String_OR_InvokeMemberName name, System.Object[] args) [0x00018] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext (Dynamitey.InvokeContext context, System.Reflection.MethodInfo method, System.Object[] parameters) [0x00023] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeExtensionMethod (System.String name, System.Reflection.MethodInfo method, System.Collections.Generic.List`1[T] parameters) [0x00034] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction (System.Type type, System.String name, System.Collections.Generic.IEnumerable`1[T] p) [0x002d8] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction (System.Type device, System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00023] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName (System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00035] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x00165] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001df] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x0019d] in <6d723dca35c642348714de4eeb24b417>:0 
[... cut ...]
  at Antmicro.Renode.Peripherals.CPU.CortexM..ctor (System.String cpuType, Antmicro.Renode.Core.Machine machine, Antmicro.Renode.Peripherals.IRQControllers.NVIC nvic, ELFSharp.ELF.Endianess endianness) [0x00007] in <68a0f1e20cbc4059bc084ef834c1d50a>:0 
  at (wrapper managed-to-native) System.Reflection.RuntimeConstructorInfo.InternalInvoke(System.Reflection.RuntimeConstructorInfo,object,object[],System.Exception&)
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <533173d24dae460899d2b10975534bb0>:0 

```

mateusz-holenko, 2021-12-02:
It looks like `mono` can't find the `libdl.so` file. Perhaps it's the `.2` postfix in the filename.
I'll look into that deeper, but a quick workaround is to add a symbolic link in `/usr/lib` - exeucting the following command worked for me:
```
sudo ln -s /usr/lib/x86_64-linux-gnu/libdl.so.2 /usr/lib/dl
```

haliskilic, 2021-12-02:
> It looks like `mono` can't find the `libdl.so` file. Perhaps it's the `.2` postfix in the filename. I'll look into that deeper, but a quick workaround is to add a symbolic link in `/usr/lib` - execucting the following command worked for me:
> 
> ```
> sudo ln -s /usr/lib/x86_64-linux-gnu/libdl.so.2 /usr/lib/dl
> ```

I executed the command, and then starting renode;

```
Exception has been thrown by the target of an invocation.
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x0001a] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.RuntimeConstructorInfo.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00086] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.RuntimeConstructorInfo.Invoke (System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at System.Reflection.ConstructorInfo.Invoke (System.Object[] parameters) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.CreateAndHandleError (System.Reflection.ConstructorInfo constructor, System.Collections.Generic.IEnumerable`1[T] attributes, System.String friendlyName, Antmicro.Renode.PlatformDescription.Syntax.IWithPosition responsibleSyntaxElement) [0x0004a] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.CreateFromEntry (Antmicro.Renode.PlatformDescription.Syntax.Entry entry) [0x00034] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessInner (System.String file, System.String source) [0x00061] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessFile (System.String path) [0x00031] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at Antmicro.Renode.PlatformDescription.UserInterface.PlatformDescriptionMachineExtensions.LoadPlatformDescription (Antmicro.Renode.Core.Machine machine, System.String platformDescriptionFile) [0x00006] in <c41dbc89274a489b81e783207c10ddb7>:0 
  at (wrapper dynamic-method) System.Object.CallSite.Target(System.Runtime.CompilerServices.Closure,System.Runtime.CompilerServices.CallSite,object,object,object)
  at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2] (System.Runtime.CompilerServices.CallSite site, T0 arg0, T1 arg1, T2 arg2) [0x0011e] in <5b415632df1f4365ae2242b1a257bb5b>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction (System.Runtime.CompilerServices.CallSite& callsite, System.Type binderType, System.Int32 knownType, Dynamitey.Internal.Optimization.InvokeHelper+LazyBinder binder, Dynamitey.InvokeMemberName name, System.Boolean staticContext, System.Type context, System.String[] argNames, System.Object target, System.Object[] args) [0x000df] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite (System.Object target, Dynamitey.InvokeMemberName name, System.Object[] args, System.String[] tArgNames, System.Type tContext, System.Boolean tStaticContext, System.Runtime.CompilerServices.CallSite& callSite) [0x0004f] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Dynamic.InvokeMemberAction (System.Object target, Dynamitey.String_OR_InvokeMemberName name, System.Object[] args) [0x00018] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext (Dynamitey.InvokeContext context, System.Reflection.MethodInfo method, System.Object[] parameters) [0x00023] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeExtensionMethod (System.String name, System.Reflection.MethodInfo method, System.Collections.Generic.List`1[T] parameters) [0x00034] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction (System.Type type, System.String name, System.Collections.Generic.IEnumerable`1[T] p) [0x002d8] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction (System.Type device, System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00023] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName (System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00035] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x00165] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001df] in <6d723dca35c642348714de4eeb24b417>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x0019d] in <6d723dca35c642348714de4eeb24b417>:0 
[... cut ...]
  at Antmicro.Renode.Peripherals.CPU.CortexM..ctor (System.String cpuType, Antmicro.Renode.Core.Machine machine, Antmicro.Renode.Peripherals.IRQControllers.NVIC nvic, ELFSharp.ELF.Endianess endianness) [0x00007] in <68a0f1e20cbc4059bc084ef834c1d50a>:0 
  at (wrapper managed-to-native) System.Reflection.RuntimeConstructorInfo.InternalInvoke(System.Reflection.RuntimeConstructorInfo,object,object[],System.Exception&)
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <533173d24dae460899d2b10975534bb0>:0 

```

PiotrZierhoffer, 2021-12-21:
It seems that your repository is in an invalid state.

There is one interesting thing here:

`Error while getting symbol from dynamic library: /tmp/renode-2501/942ed069-e533-4a39-a89f-198ab84e139b-Antmicro.Renode.translate-arm-m-le.so: undefined symbol: tlib_get_ınterrupt_vector_base`

The curious part is `tlib_get_ınterrupt_vector_base`, which should be `tlib_get_interrupt_vector_base` - with a different `i` character.

Might this be caused by some locale settings of your system? This could have been caused by opening one of the files in a text editor, or maybe by `git` doing something by itself.

mcergun, 2022-01-28:
> The curious part is `tlib_get_ınterrupt_vector_base`, which should be `tlib_get_interrupt_vector_base` - with a different `i` character.

This is caused by the localization of the terminal. I am observing same behavior in Windows commandline. If I explicitly set the localization to ENGLISH, it goes back to normal (`ı -> i`). So I don't think it's a serios problem.

mateusz-holenko, 2022-04-12:
Is this still an issue?

I've just tried running the script in a clean docker image of Ubuntu 21.10:
```
$ wget https://dl.antmicro.com/projects/renode/builds/renode-latest.linux-portable.tar.gz
$ tar -xf renode-latest.linux-portable.tar.gz
$ cd renode_1.12.0+20220407gitdd39ce44_portable
$ ./renode --console --disable-xwt
[...]
Renode, version 1.12.0.2539 (dd39ce44-202204070124)
(monitor) s @scripts/single-node/stm32f4_discovery.resc
```

and noticed no problems related to loading of a dynamic library.

PiotrZierhoffer, 2023-01-08:
As we don't see any way to reproduce this at the moment - I'm closing the issue. Feel free to open a new one if the problem persists

closed 2023-01-08 by PiotrZierhoffer