renode/renode #274: How to enable Ethernet hardware checksum while simulating stm32f7_discovery board?

minifake, 2021-11-30:
I'm trying to test lwIP communication in simulated  stm32f7_discovery board. I use Wireshark to capture the traffic and find out the TCP package's checksum is always 0x00 which causes handshake failed.

I checked the lwIP configuration in the stm32 project, it defined macro CHECKSUM_BY_HARDWARE. Then I undefined it and test again, handshake succeed with correct checksum. 

So I just wonder how to enable Ethernet hardware checksum for the renode simulator? (I really need project run normally while macro CHECKSUM_BY_HARDWARE is defined )

Thank you for your work to provide such a good platform.

mateusz-holenko, 2021-12-02:
Hi @minifake, thanks for the report.

From what I see in https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Network/SynopsysEthernetMAC.cs#L375 the ethernet card model should support generation of CRCs for outgoing frames.

Could you enable noisy logging for this peripheral and share the log?
```
(machine-0) logLevel -1 sysbus.ethernet
```

minifake, 2021-12-06:
@mateusz-holenko 

Here is the log:
```
10:17:22.6192 [NOISY] STM32F746/ethernet: Read from DMAStatusRegister
10:17:22.6193 [NOISY] STM32F746/ethernet: Write 4 to DMAStatusRegister
10:17:22.6193 [NOISY] STM32F746/ethernet: Write 0 to DMATransmitPollDemand
10:17:22.6193 [NOISY] STM32F746/ethernet: Sending frame
10:17:22.6194 [NOISY] STM32F746/ethernet: GOING TO READ FROM 2004DEC4, len=58
10:17:22.6194 [NOISY] STM32F746/sram: Invalidating memory fragment at 0x4C0A0 of size 4 bytes.
10:17:22.6194 [NOISY] STM32F746/sram: Invalidating memory fragment at 0x4C0A4 of size 4 bytes.
10:17:22.6194 [NOISY] STM32F746/sram: Invalidating memory fragment at 0x4C0A8 of size 4 bytes.
10:17:22.6194 [NOISY] STM32F746/sram: Invalidating memory fragment at 0x4C0AC of size 4 bytes.
10:17:22.6195 [NOISY] STM32F746/ethernet: Sending frame of 58 bytes.
10:17:22.6195 [NOISY] STM32F746/ethernet: Calculating checksum (mode 3).
10:17:22.6195 [DEBUG] STM32F746/ethernet: Sending packet from STM32F746, length: 62
    [EthernetPacket: SourceHwAddress=02:00:00:00:00:00, DestinationHwAddress=06:07:75:30:2a:24, Type=IpV4][IPv4Packet: SourceAddress=192.168.0.10, DestinationAddress=192.168.0.11, HeaderLength=5, Protocol=TCP, TimeToLive=255][TCPPacket: SourcePort=80, DestinationPort=57336, Flags={ack[966437230 (0x399aa96e)]|syn[0x196e,6510]}]
10:17:22.6196 [NOISY] switch1: Received frame from interface 02:00:00:00:00:00
10:17:22.6196 [NOISY] switch1: Forwarding frame to interface 06:07:75:30:2A:24
10:17:22.6196 [NOISY] STM32F746/nvic: External IRQ 77: True
10:17:22.6196 [NOISY] STM32F746/nvic: Waking up from deep sleep
10:17:22.6197 [NOISY] STM32F746/cpu: IRQ 0, value True
10:17:22.6197 [NOISY] STM32F746/cpu: Setting CPU IRQ #0 to True
10:17:22.6197 [NOISY] STM32F746/ethernet: Frame sent.
10:17:22.6197 [NOISY] STM32F746/nvic: Acknowledged IRQ 77.
10:17:22.6197 [NOISY] STM32F746/cpu: IRQ 0, value False
10:17:22.6198 [NOISY] STM32F746/cpu: Setting CPU IRQ #0 to False
10:17:22.6198 [NOISY] STM32F746/ethernet: Read from DMAStatusRegister (2)
10:17:22.6199 [NOISY] STM32F746/ethernet: Write 10000 to DMAStatusRegister
10:17:22.6199 [NOISY] STM32F746/nvic: External IRQ 77: False
10:17:22.6199 [NOISY] STM32F746/ethernet: Read from DMAStatusRegister
10:17:22.6199 [NOISY] STM32F746/nvic: Completed IRQ 77 active -> inactive.
10:17:22.6200 [NOISY] STM32F746/ethernet: Read from DMAStatusRegister
```

I can see there is `Calculating checksum (mode 3)` in the log, but the actual checksum is still 0x0.

PhilippFinke1988, 2021-12-29:
@mateusz-holenko 
I also ran into this issue, but I've found a solution. The final Ethernet frame with CRC is not created from the frame already containing the inner checksums but the initial raw data without inner checksums.

File "src\Infrastructure\src\Emulator\Peripherals\Peripherals\Network\SynopsysEthernetMAC.cs" line 393.

Change from:
`if(!Misc.TryCreateFrameOrLogWarning(this, packetData.ToArray(), out var frameWithCrc, addCrc: true))`
To:
`if(!Misc.TryCreateFrameOrLogWarning(this, frame.Bytes, out var frameWithCrc, addCrc: true))`

Could you please add this fix to the next release? I need this change, too.

PhilippFinke1988, 2021-12-30:
I created a pull request with a fix for this issue.

mateusz-holenko, 2022-02-08:
The fix for CRC calculation in SynopsysEthernetMAC is merged in: https://github.com/renode/renode/commit/bc9f0269c9693b6a83bbf6544741dce03b3ff1af.

mateusz-holenko, 2022-02-08:
Closing this issue now, but feel free to open a new one if you encounter new problems with this model in the future.

closed 2022-02-08 by mateusz-holenko

gonbbb, 2022-12-12:
@minifake 
I'm interested in using the stm32f7_discovery board on Renode to test lwip.

I know it's been some time, but if you have the resc files for this environment, could you share them with me?