renode/renode #276: Fatal error on Windows 10

haliskilic, 2021-12-04:
Hi,

I'm trying to start the demo project. When I execute the demo project command, I see this fatal error. I shared all information.

**Renode Information:**

`Renode, version 1.12.0.6130 (e07f2ae3-202112040323)`

**Device specifications:**

```
Device name	DESKTOP-ED81ASU
Processor	Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz   3.40 GHz
Installed RAM	8,00 GB
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
```

**Windows specifications:**

```
Edition	Windows 10 Pro
Version	20H2
Installed on	‎28.‎11.‎2020
OS build	19042.1348
Experience	Windows Feature Experience Pack 120.2212.3920.0
```

**Steps:**
Step 1 -> start renode with console

`C:\>renode`

Step 2 -> start stm32f4_discovery.resc with monitor

`s @scripts/single-node/stm32f4_discovery.resc`

Result -> fatal error occurs !

**Console Output:**
```

15:48:38.6631 [INFO] Loaded monitor commands from: C:\Program Files\Renode\scripts/monitor.py
15:48:42.1624 [INFO] Including script: C:\Program Files\Renode\scripts\single-node\stm32f4_discovery.resc
15:48:42.1928 [INFO] System bus created.
Fatal error:
Exception has been thrown by the target of an invocation.
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at Antmicro.Renode.PlatformDescription.CreationDriver.CreateAndHandleError(ConstructorInfo constructor, IEnumerable`1 attributes, String friendlyName, IWithPosition responsibleSyntaxElement)
   at Antmicro.Renode.PlatformDescription.CreationDriver.CreateFromEntry(Entry entry)
   at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessInner(String file, String source)
   at CallSite.Target(Closure , CallSite , Object , Object , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2](CallSite site, T0 arg0, T1 arg1, T2 arg2)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction(CallSite& callsite, Type binderType, Int32 knownType, LazyBinder binder, InvokeMemberName name, Boolean staticContext, Type context, String[] argNames, Object target, Object[] args)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite(Object target, InvokeMemberName name, Object[] args, String[] tArgNames, Type tContext, Boolean tStaticContext, CallSite& callSite)
   at Dynamitey.Dynamic.InvokeMemberAction(Object target, String_OR_InvokeMemberName name, Object[] args)
   at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext(InvokeContext context, MethodInfo method, Object[] parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction(Type type, String name, IEnumerable`1 p)
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction(Type device, String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName(String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.TryExecuteScript(String filename, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Commands.IncludeFileCommand.Run(ICommandInteraction writer, PathToken path)
   at Antmicro.Renode.UserInterface.Commands.StartCommand.Run(ICommandInteraction writer, PathToken path)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at Antmicro.Renode.UserInterface.Monitor.RunCommand(ICommandInteraction writer, Command command, IList`1 parameters)
[... 44 lines cut ...]
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction(Type device, String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName(String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.TryExecuteScript(String filename, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Commands.IncludeFileCommand.Run(ICommandInteraction writer, PathToken path)
   at Antmicro.Renode.UserInterface.Commands.StartCommand.Run(ICommandInteraction writer, PathToken path)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at Antmicro.Renode.UserInterface.Monitor.RunCommand(ICommandInteraction writer, Command command, IList`1 parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)
```

PiotrZierhoffer, 2021-12-21:
Hi @haliskilic , thanks for the report.

Did you build Renode yourself or did you use one of our nightly packages?

Can you please try this version? https://dl.antmicro.com/projects/renode/builds/renode-latest.zip

If it works for you, then the problem should be with the building process. If it doesn't, then we'll investigate the runtime further.

mcergun, 2022-01-28:
Hi @PiotrZierhoffer,
Even though I am not the original reporter, I am facing the same issue as well.

Here's what I've tried so far:

Try different renode versions:
- 1.11.0.8104
- 1.12.0.6118 (latest.msi from build repo)
- 1.12 from Github Releases

None of them works and gives a similar error (I can confirm that the error is exactly the same for 1.12.0.6118)

I've checked my system for a valid NET 4.7 installation and confirmed that it is there.
I've enabled and disabled Advanced .NET services from Windows Features.
I've installed .NET 4.8 with the offline installer,

Nothing seems to work.

Here's what I do to get the same error:
```
(monitor) include @scripts/single-node/stm32f4_discovery.resc
```

However, `stm32f4_discovery` is not mandatory the get the error. Every script gives the same error (or download errors as my development computer is not connected to the internet)

renode crash logs list the same exception file, which is why I am not including them in the report.

tuygunol, 2023-02-23:
The same thing had happended to us. We got the same error log provided in the first entry.

The step that solved our problem is just change the time format of PC to English (United States) from Control Panel -> Region settings.

Hope this step will fix everyone else's problem.

sezginkarapinar, 2023-11-13:
> Hi @PiotrZierhoffer ,
> 
> The same thing had happended to us. We got the same error log provided in the first entry.
> 
> The step that solved our problem is just change the time format of PC to English (United States) from Control Panel -> Region settings.
> 
> Hope this step will fix everyone else's problem.

It worked for me. Thank you so much.