renode/renode #279: Which kind of situations will cause UsageFault_Handler for Cotex-

minifake, 2021-12-06:


minifake, 2021-12-06:
I found out the problem will 100% happen when using `float_t` to do type conversion.
```
  uint32_t   testUint = 0;
  float_t       testFloat;

  testFloat = (float_t) testUint;      //after this execution, indicated into UsageFault_Handler
```

PiotrZierhoffer, 2021-12-29:
Hi @minifake . 

Hardware FPU is generally supported, and we should investigate why does it fail in your case.

First of all, can you please post your .repl file? Especially the way you define the CPU and NVIC is interesting.

FPU is usually enabled by a write to NVIC in Renode. You can also try to forcefully enable FPU support and then try if it fails for you.

Just after you load the script, but before you start the emulation, run
```
cpu FpuEnabled true
```

Please let me know if it changes anything in your case.