renode/renode #281: does Renode's RCC works on STM32F407VGTx Model ?

jjpark78, 2021-12-17:
i am testing a simple firmware developed by rust. 
and seems pretty cool ~~. 

but when i  added some delay routine using RCC & Timer into code, 
Renode Stop to working. looks waiting for timer event. but not triggered. 

but on real hardware the code is fine.  
```rust
#![no_main]
#![no_std]

use panic_probe as _;
use rtt_target::{rtt_init_print, rprintln};

use crate::hal::{
    block,
    pac,
    prelude::*,
    serial::Serial,
    serial::config::Config,
    timer::Timer,
};

use stm32f4xx_hal as hal;
use cortex_m_rt::entry;
use core::fmt::Write;

#[entry]
fn main() -> ! {
    rtt_init_print!(); // You may prefer to initialize another way
    rprintln!("Hello, world!");

    let hp = pac::Peripherals::take().unwrap();
    let cp = cortex_m::Peripherals::take().unwrap();

    let rcc = hp.RCC.constrain();
    let clocks = rcc.cfgr.use_hse(8.mhz()).freeze();

    let gpiod = hp.GPIOD.split();
    let gpioa = hp.GPIOA.split();

    // Initialize on-board LEDs
    let mut led_bit5 = gpiod.pd12.into_push_pull_output();
    let mut led_bit6 = gpiod.pd13.into_push_pull_output();
    let mut led_bit7 = gpiod.pd14.into_push_pull_output();
    let mut led_bit8 = gpiod.pd15.into_push_pull_output();    // setup uart & serial

    let uart2_tx = gpioa.pa2.into_alternate();
    let uart2_rx = gpioa.pa3.into_alternate();

    let mut timer = Timer::syst(cp.SYST, &clocks).start_count_down(1.hz());
    let mut tx = Serial::usart2(
        hp.USART2,
        (uart2_tx, uart2_rx),
        Config::default().baudrate(9600.bps()),
        clocks,
    ).unwrap(); // Constrain clock registers

    writeln!(tx, "USART2 Initailed");
    // Configure clock to 168 MHz (i.e. the maximum) and freeze it
    let mut counter = 0;
    loop {
        block!(timer.wait()).unwrap();
        led_bit8.set_low();
        led_bit5.set_high();
        block!(timer.wait()).unwrap();
        led_bit5.set_low();
        led_bit6.set_high();
        block!(timer.wait()).unwrap();
        led_bit6.set_low();
        led_bit7.set_high();
        block!(timer.wait()).unwrap();
        led_bit7.set_low();
        led_bit8.set_high();

        rprintln!("Led Blink Cycle No {}, done", counter);
        writeln!(tx, "Led Blink Cycle No {}, done", counter);
        counter = counter + 1;
    }
}
```

First USART Message "USART2 Initailed" is shown at Uart Analyzer. ye~~
But renode enter to infinite loop on block! macro, cause timer is not working properly. (real hardware has no problem. working out of box)

so remove all blocks! macro, 
Second writeln! macro is working. 

is this bug of renode ?
i do want to know if there is some of proper timer initialize steps

thanks.

mateusz-holenko, 2021-12-29:
Hi @jjpark78, thanks for reporting the issue.

Could you share the compiled binary and exact scripts/platforms you use? This would make it easier to investigate the problem.

jjpark78, 2021-12-30:
target : thumbv7em-none-eabihf
[jltech-rust-firmware-tutorial01.zip](https://github.com/renode/renode/files/7791072/jltech-rust-firmware-tutorial01.zip)

target : thumbv7em-none-eabi
[jltech-rust-firmware-tutorial01.zip](https://github.com/renode/renode/files/7791073/jltech-rust-firmware-tutorial01.zip)