renode/renode #282: Host network connection (TAP) on Windows

PhilippFinke1988, 2021-12-29:
I'm currently looking for a solution to emulate the firmware of all our products (mostly ARM based) on Windows and Renode seems to fulfill most of our requirements. The only feature that is missing is the host network connection (TAP) under Windows. As we need to connect to our devices with some company owned tools that are only available under Windows this is mandatory feature for us.

Are there any plans to intergrate this feature in the near future?

PiotrZierhoffer, 2021-12-29:
Hi @PhilippFinke1988,

TAP support for Windows is currently in the works. We have a working support for this internally, but we are still processing the code before public release.

We’d be happy to talk more about your use case, so please reach out to contact@antmicro.com!

PhilippFinke1988, 2021-12-29:
Hi @PiotrZierhoffer,

thanks for the fast reply. That are great news. When is this scheduled for release?

PiotrZierhoffer, 2021-12-29:
You should expect it early next year - the solution is prepared, it just requires some polishing. While I'm not sure if it will be a part of the next numbered release, we provide nightly builds at builds.renode.io which can be used as soon as we release the code.

PhilippFinke1988, 2021-12-29:
Thank you very much. I'll keep an eye on your change history.

Best regards and a happy new year!

PiotrZierhoffer, 2022-03-29:
The current version, while not yet on master, is available here: https://github.com/renode/renode/tree/tun_tap_on_windows

It requires openvpn to be installed on host. Please note it's work in progress.

PiotrZierhoffer, 2023-01-08:
This is actually long merged. Our documentation needs to be updated to reflect this, but the feature is available - closing the issue.

closed 2023-01-08 by PiotrZierhoffer