renode/renode #286: Unhandled exception in LLVMDisassembler.

lcavalli, 2022-01-11:
In renode nightly builds on Windows 10 an unhandled exception is rised at startup:

```
C:\>renode
Fatal error:
L'inizializzatore di tipo di 'Antmicro.Renode.Utilities.TypeManager' ha generato un'eccezione.
   in Antmicro.Renode.Utilities.TypeManager.get_Instance()
   in Antmicro.Renode.UserInterface.Monitor..ctor()
   in Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun)
   in Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1()
   in System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   in System.Threading.ThreadHelper.ThreadStart()
Inner exception:
Il metodo 'TryDisassembleInstruction' del tipo 'Antmicro.Renode.Disassembler.LLVM.LLVMDisassembler' dell'assembly 'LLVMDisassembler, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null' non ha un'implementazione.
   in System.RuntimeTypeHandle.GetTypeByName(String name, Boolean throwOnError, Boolean ignoreCase, Boolean reflectionOnly, StackCrawlMarkHandle stackMark, IntPtr pPrivHostBinder, Boolean loadTypeFromPartialName, ObjectHandleOnStack type)
   in System.RuntimeTypeHandle.GetTypeByName(String name, Boolean throwOnError, Boolean ignoreCase, Boolean reflectionOnly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean loadTypeFromPartialName)
   in System.Type.GetType(String typeName)
   in Antmicro.Renode.Utilities.TypeManager.GetTypeWithLazyLoad(String name, String assemblyFullName, String path)
   in Antmicro.Renode.Utilities.TypeManager.AnalyzeAssembly(String path, Boolean bundled, Boolean throwOnBadImage)
   in Antmicro.Renode.Utilities.TypeManager.ScanInner(String path, Boolean recursive)
   in Antmicro.Renode.Utilities.TypeManager.Scan(String path, Boolean recursive)
   in Antmicro.Renode.Utilities.TypeManager..cctor()

Eccezione non gestita: System.TypeInitializationException: L'inizializzatore di tipo di 'Antmicro.Renode.Utilities.TypeManager' ha generato un'eccezione. ---> System.TypeLoadException: Il metodo 'TryDisassembleInstruction' del tipo 'Antmicro.Renode.Disassembler.LLVM.LLVMDisassembler' dell'assembly 'LLVMDisassembler, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null' non ha un'implementazione.
   in System.RuntimeTypeHandle.GetTypeByName(String name, Boolean throwOnError, Boolean ignoreCase, Boolean reflectionOnly, StackCrawlMarkHandle stackMark, IntPtr pPrivHostBinder, Boolean loadTypeFromPartialName, ObjectHandleOnStack type)
   in System.RuntimeTypeHandle.GetTypeByName(String name, Boolean throwOnError, Boolean ignoreCase, Boolean reflectionOnly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean loadTypeFromPartialName)
   in System.Type.GetType(String typeName)
   in Antmicro.Renode.Utilities.TypeManager.GetTypeWithLazyLoad(String name, String assemblyFullName, String path)
   in Antmicro.Renode.Utilities.TypeManager.AnalyzeAssembly(String path, Boolean bundled, Boolean throwOnBadImage)
   in Antmicro.Renode.Utilities.TypeManager.ScanInner(String path, Boolean recursive)
   in Antmicro.Renode.Utilities.TypeManager.Scan(String path, Boolean recursive)
   in Antmicro.Renode.Utilities.TypeManager..cctor()
   --- Fine della traccia dello stack dell'eccezione interna ---
   in Antmicro.Renode.Utilities.TypeManager.get_Instance()
   in Antmicro.Renode.UserInterface.Monitor..ctor()
   in Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun)
   in Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1()
   in System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   in System.Threading.ThreadHelper.ThreadStart()
```

It seems it has been introduced in `renode_1.12.0+20211215git2a79488d.msi`, since `renode_1.12.0+20211214gitc01040f6.msi` is not effected.

PiotrZierhoffer, 2022-02-01:
Thank you for the report, we'll look into this soon.

Can you please verify if the latest nightly build also exposes this issue?

lcavalli, 2022-02-02:
Hi.
No, it seems that current nightly build doesn't have this issue and runs as expected. Closing.

closed 2022-02-02 by lcavalli