renode/renode #288: Error E25: Could not find suitable constructor

csogawa, 2022-01-17:
I'm trying to create a model for the STM32U575 processor. I copied the stm32l552.repl which is an earlier predecessor, changed the addresses as necessary and loaded one of the sample STM project. The firmware fails the code that waits for the PLL to lock so I tried to add the RCC defined for the STM32F4, i.e., Miscellaneiou.STM32F4_RCC.cs. However, when I add this to the .repl I get the following error message:

(machine-0) machine LoadPlatformDescription @platforms/cpus/stm32u575.repl 
There was an error executing command 'machine LoadPlatformDescription @E:\cygwin64\home\CRAI
G.OGAWA\renode\platforms\cpus\stm32u575.repl'
Error E25: Could not find suitable constructor for type 'Antmicro.Renode.Peripherals.Miscell
aneous.STM32F4_RCC'.
Constructor selection report:
Considering ctor Antmicro.Renode.Peripherals.Miscellaneous.STM32F4_RCC with the following pa
rameters: [Antmicro.Renode.Core.Machine, Antmicro.Renode.Peripherals.Timers.STM32F4_RTC].
  Parameter 'machine' of type 'Antmicro.Renode.Core.Machine' filled with default value = 'ma
chine-0'.
  Could not find corresponding attribute for parameter 'rtcPeripheral' of type 'Antmicro.Ren
ode.Peripherals.Timers.STM32F4_RTC' and it is not a default parameter. Rejecting constructor
.
At E:\cygwin64\home\CRAIG.OGAWA\renode\platforms\cpus\stm32u575.repl:100:6:
rcc: Miscellaneous.STM32F4_RCC @ sysbus 0x46020C00
     ^
(machine-0) 

I'm guessing that something defined for this class is preventing this from loading properly. I'm planning on starting with the STM32F4_RCC class to implement one for the STM32U5 but do not know what I should be looking for to resolve this issue.

I've attached the .repl.
[stm32u575.repl.txt](https://github.com/renode/renode/files/7884925/stm32u575.repl.txt)

Is there a way to implement this with a python script?

csogawa, 2022-01-17:
As I reviewed the initial posting I see why the constructor is failing, i.e., the RTC class type is not STM32F4_RTC. I will proceed with the new STM32U5_RCC class supporting this RTC.

But would still be interested in knowing if it is possible to do the same thing with a python script.

closed 2022-01-17 by csogawa

mateusz-holenko, 2022-01-18:
You are right, currently `STM32F4_RCC` requires `STM32F4_RTC` as a parameter. Do you know how similar the models of RCC and RTC for STM32U5 are compared to those for STM32F4? Perhaps they could generalized to fit  both platforms?

As for using python - you can write a model of peripheral in python. Take a look at an example: https://github.com/renode/renode/blob/master/platforms/cpus/stm32f0.repl#L147-L162.

csogawa, 2022-01-18:
The RCC is not very similar to between the two devices so I’ve started implementing one for the STM32U5. There are a lot of additional registers defined for STM32U5 and only a few common ones. Once I’ve completed this I can compare to see if a base class can be extracted.

Thank you for the pointer to the example python implementation. Since I’ve started the C# implementation I’ll go ahead and finish it out but I have a lot of additional peripherals that can be implemented in python.