renode/renode #289: Failed to build from source on Window 11

vietmaiquoc, 2022-01-19:
Hi, 

I trying to compile Renode from source on my Window11 PC and encounter the errors as in the log file
[Renode_build_log.txt](https://github.com/renode/renode/files/7894266/Renode_build_log.txt)

Please help me take a look.

thanks
Viet
Ion Mobility Singapore

vietmaiquoc, 2022-01-19:
for more detail, i actually got the issue : 
[image]

PiotrZierhoffer, 2022-01-21:
Hi @vietmaiquoc 

It seems like you are lacking a toolchain for some reason.

Can you please verify if you reproduced all the steps from https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html ? Specifically if you have the toolchain available in `%PATH%`?

xobs, 2022-02-28:
I use this powershell script to build on Windows. Note that I use `musl.cc` for mingw on Windows:

```powershell
param (
    [Boolean]$submodules = $true,
    [switch]$debug = $false,
    [switch]$clean = $false
)

$env:PATH += ";C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin"

# This should be the path to a "bin/" directory of the musl.cc toolchain,
# available for download from https://musl.cc/
$env:PATH += ";E:\Software\musl.cc\x86_64-w64-mingw32-native\bin\"

$target = "ReleaseWindows"
$target_path = "Release"
if ($debug) {
    $target = "DebugWindows"
    $target_path = "Debug"
}

# We should only do this if "-s" is passed, or if submodules
# are not initialized
# if ($submodules) {
#     git submodule update --init --recursive
# }

# It's unclear why this is a separate repository.
git -C lib clone https://github.com/renode/renode-resources.git resources
git -C lib/resources fetch
git -C lib/resources checkout origin/master

# Update references to Xwt
(Get-Content .\lib\termsharp\TermSharp.csproj)|
    ForEach-Object {$_ -replace """xwt\\Xwt\\Xwt.csproj", """..\xwt\Xwt\Xwt.csproj"} |
    Set-Content .\lib\termsharp\TermSharp.csproj

# Build CCTask, which is used to run subtasks
MSBuild /p:Configuration=Release /p:Platform="Any CPU" .\lib\cctask\CCTask.sln

# Copy the properties file, which contains various build environment settings
if (-not (Test-path output\properties.csproj)) {
    if (-not (Test-path output)) {
        New-Item -Type Directory output
    }
    Copy-Item src\Infrastructure\src\Emulator\Cores\windows-properties.csproj output\properties.csproj
}

# Run the "clean" task if the user requested
if ($clean) {
    MSBuild /p:Configuration=ReleaseWindows /t:Clean Renode.sln
    MSBuild /p:Configuration=DebugWindows /t:Clean Renode.sln
    return
}

MSBuild /p:Configuration=$target Renode.sln

# copy llvm library
Copy-Item src/Infrastructure/src/Emulator/LLVMDisassembler/bin/$target_path/libLLVM.* output/bin/$target_path
```