renode/renode #290: Create a tutorial describing how to configure IPv4 network with DHCP to work with Renode

mateusz-holenko, 2022-02-01:
Configuring the environment requires executing many commands - it would be great to have it written down in one place.

The topic has been discussed in #148 and some implementation proposed in https://github.com/renode/renode-infrastructure/pull/29, but I believe it would be better to have it in a form of a tutorial instead of a group of new host-dependent commands in Renode.

jensschroer, 2022-03-03:
I think it was the right decision to rather write this as a piece in the tutorial section than using the code directly.
For help, I will just write up  the steps that I remember (note, this is based on Ubuntu/Linux).

The basic idea is to create the following architecture:
```mermaid
graph LR;
 subgraph Renode;
  R1(Machine) --- RS[switch1];
  RS --- RT[tap];
 end;
 subgraph Linux;
 T[tap0] --- BR[renode-br0];
 D(dnsmasq) --- BR;
 end;
 RT ---- T;
```
We only need to set up the basic configuration of the tap and bridge devices on the Linux machine.
In this way, the renode machines can send DHCP request to the Linux host, that runs a dedicated `dnsmasq` server.
Package routing control is done with the help of `ufw`.

## Required Packages
- iproute2 (for the ip tool)
- dnsmasq (DHCP)
- ufw (for configuring the traffic to/from the network interface)
 

## Steps
### In Renode
Setup a wired network in the repl file, containing at least a switch and define a tap interface.
```
emulation CreateSwitch "switch1"
emulation CreateTap "tap0" "tap"
connector Connect host.tap switch1
```
### Linux
Following commands usually need to be run as sudo. 

#### Tap and Bridge device

```
# Create the bridge device
ip link add name renode-br0 type bridge

# Set the IP and netmask for the bridge
ip addr add 192.168.1.1/24 brd + dev renode-br0
ip link set renode-br0 up
```

For the next step Renode needs to be running, as it will create the `tap0` interface.

```
# Connect bridge and tap0
ip link set tap0 master renode-br0
ip link set tap0 up
```
#### Configure package routing / firewall
For packages to flow between Renode and the Linux user space, package routing has to be configured.
The easiest way I found was by using `ufw` - the uncomplicated firewall.

The following commands enable the traffic between Renode and Linux:
```
ufw allow in on renode-br0
ufw allow out on renode-br0
```

#### DHCP Server
I used `dnsmasq` for this. Other DHCP servers should also be possible.
This command starts the server on the bridge interface `renode-br0` with an dhcp-range of `192.168.1.2 - 192.168.2.10`.
```
dnsmasq --log-queries --no-hosts --no-resolv --leasefile-ro --interface=renode-br0 -p0 --log-dhcp --dhcp-range=192.168.1.2,192.168.1.10 
```
#### Tearing it all down
First, shutdown the DHCP server.
Then revert the settings made in the previous sections, the following commands need to be executed in order:
```
# Undo the firewall rules
ufw delete allow in on renode-br0
ufw delete allow out on renode-br0

# Remove the bridge and shutdown tap
ip link delete renode-br0
ip link set tap0 down
```

I hope this helps for writing the tutorial. It might be good if someone else could just try these steps themself, just to make sure I did not miss anything.

ekawahyu, 2023-06-01:
@jensschroer I can confirm that those steps work on ZorinOS 16.2 (Ubuntu 20.04.5 LTS based)

ekawahyu, 2023-06-01:
I would like to share the following setup that works for me. With this setup, Renode machine can join your local network and get IP address from the existing DHCP server. Ideally, the network switch/router has multiple LAN ports, but they are intentionally removed for clarity:

```mermaid
graph LR;
 subgraph Renode;
  R1(Machine) --- RS[switch1];
  RS --- RT[tap];
 end;
 subgraph Linux;
 T[tap0] --- BR[br0];
 E[eth0] --- BR;
 end;
 subgraph Network Switch/Router;
  D1(DHCP Server) --- L1[lan1];
 end;
 RT ---- T;
 L1 ---- E;
```

## Steps

### In Renode

Setup a wired network in the repl file, containing at least a switch and define a tap interface.
```
emulation CreateSwitch "switch1"
emulation CreateTap "tap0" "tap"
connector Connect host.tap switch1
```

### In Linux

Following commands usually need to be run as sudo. Please make sure Renode is already running.

Ethernet, Tap, and Bridge device setup
```
ip addr flash dev eth0
ip link add name br0 type bridge
ip link set eth0 master br0
ip link set tap0 master br0
ip link set up dev eth0
ip link set up dev tap0
ip link set up dev br0
dhclient -v br0
```

Enable traffic between Renode and Linux:
```
ufw allow in on br0
ufw allow out on br0
```

I hope this helps anyone who would like to get their Renode machine connected to their existing network.

ekawahyu, 2023-06-02:
Configuring TAP device and DHCP on macOS (Monterey, tested) is a bit tricky, because Apple has made it more and more difficult to load system extensions with each new version of macOS. They have also announced that in "a future version" of macOS, you will not be able to use system extensions at all (reference: https://tunnelblick.net/cTunTapConnections.html)

TAP device can be created with Tunnelblick's Tap system extension. So the first prerequisite is installing Tunnelblick with its system extension. Detail guide is provided here: https://tunnelblick.net/cKextsInstallation.html

Once Tunnelblick system extension is properly installed, executing the following command to load it:

```
/Applications/Tunnelblick.app/Contents/Resources/openvpnstart loadKexts 2
```

Detail explanation on how to use openvpnstart and its parameter is provided here: https://github.com/Tunnelblick/Tunnelblick/issues/703

Next prerequisite, the `dnsmasq`, can be installed using Homebrew:

```
brew install dnsmasq
```

The slightly modified architecture from original post by @jensschroer:

```mermaid
graph LR;
 subgraph Renode;
  R1(Machine) --- RS[switch1];
  RS --- RT[tap];
 end;
 subgraph macOS;
 T[tap0] --- BR[bridge1];
 D(dnsmasq) --- BR;
 end;
 RT ---- T;

```

Ok, so let's get the setup going!

## Steps In Renode

Setup a wired network in the repl file, containing at least a switch and define a tap interface.
```
emulation CreateSwitch "switch1"
emulation CreateTap "tap0" "tap"
connector Connect host.tap switch1
```

## Steps In macOS

Following commands need to be run as sudo. Please make sure Renode is already running.

### Setup

TAP and Bridge device setup
```
ifconfig bridge1 create
ifconfig bridge1 up
ifconfig bridge1 192.168.10.1/24
ifconfig bridge1 addm tap0
```

This command starts the server on the bridge interface `bridge1` with an dhcp-range of 192.168.10.2 - 192.168.10.10.
```
dnsmasq --log-queries --no-hosts --no-resolv --leasefile-ro --interface=bridge1 -p0 --log-dhcp --dhcp-range=192.168.10.2,192.168.10.10
```

### Cleanup
```
ifconfig bridge1 destroy
/Applications/Tunnelblick.app/Contents/Resources/openvpnstart unloadKexts 2
```

## Troubleshooting

- When `dnsmasq` throws an error, such as socket address or port is in use, please make sure that there is no other DNS or DHCP services running. Apps like Docker and Multipass may use port 53(DNS) and/or 67(DHCP), and also please make sure Internet Sharing is disabled.
- Persistent TAP interface is not supported on macOS, so keep adding `tap0` to `bridge1` manually whenever you quit and relaunch Renode.

## Scripting

TODO: Scripting can simplify the setup and cleanup process. Reference:

http://www.mulle-kybernetik.com/weblog/2013/bridging_a_wireless_and_an_eth.html

jensschroer, 2023-06-02:
@ekawahyu Thank you for the confirmation and the additional setup options.

I have created a small bash script that automates the task under Ubuntu.
As a possible improvement one could change it to take the address range as a command line argument, but this script is sufficient for me for now.
```bash
#!/bin/bash

echo "Creating and setting up the bridge"
# Create the bridge device
ip link add name renode-br0 type bridge

# Set the IP and netmask for the bridge
ip addr add 192.168.67.1/24 brd + dev renode-br0
ip link set renode-br0 up

echo "Creating the tap interface"
# Create tap interface
ip tuntap add tap0 mode tap

# Connect bridge and tap0
ip link set tap0 master renode-br0
ip link set tap0 up

echo "Configuring firewall"
# Enable firewall package routing
ufw allow in on renode-br0
ufw allow out on renode-br0

cleanup() {
  echo ""
  echo "Cleaning up"
  # Undo the firewall rules
  echo "Removing firewall rules"
  ufw delete allow in on renode-br0
  ufw delete allow out on renode-br0

  echo "Removing bridge and shutting down tap"
  # Remove the bridge and shutdown tap
  ip link delete renode-br0
  ip link set tap0 down

  echo "Cleanup completed"
}

# Setting up signal handling
trap "exit" INT TERM ERR
trap "cleanup; kill 0" EXIT

# Starting DHCP Server
echo "************************"
echo "* Starting DHCP Server *"
echo "* Stopping with Ctrl-C *"
echo "************************"

dnsmasq --log-queries \
        --no-hosts \
        --no-resolv \
        --leasefile-ro \
        -d \
        --interface=renode-br0 \
        -p0 --log-dhcp \
        --dhcp-range=192.168.67.2,192.168.67.10
```

ekawahyu, 2023-06-03:
@jensschroer In the script, I added `ip link delete tap0` during cleanup to mute: `ioctl(TUNSETIFF): Device or resource busy`. Other than that, everything seems to work great! I can confirm that it works on both Ubuntu 20.04.5 LTS and within Windows 11 WSL2. Thanks!

Afreeinsain, 2025-01-16:
hello, I followed the tutorials above, and I have question now.  My .resc file is below, what should i do to make renode(stm32) get the ipv4, and the get udp communication with the host?
thank you very much if you can help me (:

mach create "A"
machine LoadPlatformDescription @platforms/boards/stm32f4_discovery-kit.repl

showAnalyzer sysbus.usart2

emulation CreateTap "tap0" "tap"
emulation CreateSwitch "switch1"
connector Connect host.tap switch1
connector Connect sysbus.ethernet switch1

sysbus LoadELF "/home/zy/satellite/rodos/tutorials/10-first-steps/tst"
machine SetAdvanceImmediately true
start

jensschroer, 2025-01-16:
@Afreeinsain In order to work with this setup, your firmware needs to request an IP address via DHCP. 
Have you configured it accordingly?

Afreeinsain, 2025-01-16:
I am a newbie. I am using machine LoadPlatformDescription @platforms/boards/stm32f4_discovery-kit.repl. I can't find how to set up the DHCP service in the firmware. Do you know how to do it? Or is there any related tutorial?

jensschroer, 2025-01-16:
@Afreeinsain that depends on your tech stack. Check the documentation of your embedded software. 
Are you using Zephyr? In there it should be fairly straight forward from what I remember.

DenSandu, 2026-02-18:
Hi, I have tried following this tutorial in a devcontainer but I can't find out why it's not working
This is what I did
I built the zephyr.elf file for the b_u585i_iot02a with the enc28j60 ethernet module
```
&spi1 {
    eth0: ethernet@0 {
        status = "okay";
        reg = <0>;
        compatible = "microchip,enc28j60";
        int-gpios = <&gpioc 1 GPIO_ACTIVE_LOW>;
        spi-max-frequency = <10000000>;
        local-mac-address = [ 00 80 00 01 02 03 ];
        full-duplex;
    };
};
```
I made a .resc file that loaded the .repl file containing: 
```
spi1: SPI.STM32H7_SPI @ sysbus 0x40013000
    IRQ->nvic0@59

eth0: Network.ENC28J60 @ spi1
    -> gpioc@1
```
the .resc file does also have the code to create the switch and the tap interface
```
emulation CreateSwitch "switch1"
connector Connect sysbus.spi1.eth0 switch1
emulation CreateTap "tap0" "tap"
connector Connect host.tap switch1
```

in a second terminal I did 
```
ip link add name renode-br0 type bridge

ip addr add 192.168.1.1/24 brd + dev renode-br0
ip link set renode-br0 up

ip link set tap0 master renode-br0
ip link set tap0 up
ufw allow in on renode-br0
ufw allow out on renode-br0
```
in a third terminal I did
```
dnsmasq --no-daemon --interface=renode-br0 --bind-interfaces --dhcp-range=192.168.1.2,192.168.1.10
```

and then I started the simulation of the board.
But when doing the dhcp request, it doesn't arrive to the dnsmasq
```
I: ethernet@0: Initialized
Waiting for network to be ready...
Unable to connect to network
```

I tried also seeing the component with ```peripherals``` command and I get
```
  ?       
  ??? spi1 (STM32H7_SPI)
  ?   ?   <0x40013000, 0x400133FF>
  ?   ?   
  ?   ??? eth0 (ENC28J60)
  ?   
```

So I think there might be problems with the interrupts
These are the full logs with also debug mode ```logLevel -1 sysbus.spi1.eth0```
```
[... 375 lines cut ...]
16:17:36.5255 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [6] when writing value 0x40. Tags: OVRIE (0x1).
16:17:36.5263 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [7] when writing value 0x80. Tags: CRCEIE (0x1).
16:17:36.5270 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [8] when writing value 0x100. Tags: TIFREIE (0x1).
16:17:36.5274 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [9] when writing value 0x200. Tags: MODFIE (0x1).
16:17:36.5279 [DEBUG] spi1.eth0: Read from 0x1F, bank 2.
16:17:36.5295 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [5] when writing value 0x20. Tags: UDRIE (0x1).
16:17:36.5300 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [6] when writing value 0x40. Tags: OVRIE (0x1).
16:17:36.5303 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [7] when writing value 0x80. Tags: CRCEIE (0x1).
16:17:36.5307 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [8] when writing value 0x100. Tags: TIFREIE (0x1).
16:17:36.5311 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [9] when writing value 0x200. Tags: MODFIE (0x1).
16:17:36.5316 [DEBUG] spi1.eth0: Write to 0x1F, bank 2, value 0x3.
16:17:36.5363 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [5] when writing value 0x20. Tags: UDRIE (0x1).
16:17:36.5369 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [6] when writing value 0x40. Tags: OVRIE (0x1).
16:17:36.5374 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [7] when writing value 0x80. Tags: CRCEIE (0x1).
16:17:36.5378 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [8] when writing value 0x100. Tags: TIFREIE (0x1).
16:17:36.5382 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [9] when writing value 0x200. Tags: MODFIE (0x1).
16:17:36.5399 [DEBUG] spi1.eth0: Read from 0xA, bank 3.
16:17:36.5463 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [5] when writing value 0x20. Tags: UDRIE (0x1).
16:17:36.5468 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [6] when writing value 0x40. Tags: OVRIE (0x1).
16:17:36.5471 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [7] when writing value 0x80. Tags: CRCEIE (0x1).
16:17:36.5474 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [8] when writing value 0x100. Tags: TIFREIE (0x1).
16:17:36.5478 [WARNING] spi1: Unhandled write to offset 0x10. Unhandled bits: [9] when writing value 0x200. Tags: MODFIE (0x1).
16:17:36.5481 [DEBUG] spi1.eth0: BitSet to 0x1F, bank 3, bits to set: 2.
16:17:36.5760 [INFO] usart1: [host: 0.19ks (+0.19ks)|virt: 0.11s (+0.11s)] I: ethernet@0: Initialized
```