renode/renode #292: Monodevelop can't build Renode anymore

grifcj, 2022-02-02:
Building and debugging Renode with Monodevelop has stopped working for me. I was using it to do development on/with Renode just a few weeks ago. Command line build via ./build.sh works. Below is what I'm seeing. Seems to be an issue with updating Nuget packages for the Migrant and Termsharp dependencies.

[image]
[image]

Things I've tried:
* Rebuilds/cleans, the errors just persist
* Deleting all bins/objects
* Fresh repo clone 
* Errors persist even for older commits that used to work, e.g. checking out work several hundred commits back
* Re-installed mono and monodevelop
* Use a monodevelop docker container as way to check for environment problems

One thing I haven't tried is setting up a new Ubuntu virtual machine and building in that. Figured the docker would have been good enough check against that though.

grifcj, 2022-02-05:
Appears Monodevelop has issue regarding restoring packages. Doesn't seem to like this line in Migrant.csproj.

```
  <Import Project="$(MSBuildBinPath)\Microsoft.CSharp.targets" Condition="$(NET) != 'true'"/>
```

I would try an older version of monodevelop if I could determine where to find one. Install directions only seem to indicate how to install latest version from package manager. Tried building monodevelop from source, but all three tags I tried building didn't compile with the instructions on the github. The repo is marked as unmaintained since 2020.

Also tried building a docker image with 18.04, but it pulled the same version of monodevelop as 20.04.

Some alternatives for now may be: 

1) sdb for command line debugging. Haven't tried it
2) Jetbrains Rider. Built and debug Renode just fine. Might just do this; course it costs a bit of money. 
3) Use work computer with Windows environment and try Visual Studio?

Would be interested to know what the build/dev environment is for core dev team. You running monodevelop on some flavor of linux? What exact versions?

grifcj, 2022-02-06:
sdb was able to launch renode in debugger. Last release 1.6 is from 2018. I used 1.2.7354 which has bug in passing args to renode (apparently fixed in 1.3).

Going to close this as sdb and Jetbrains being workable options in lieu of MonoDevelop.

closed 2022-02-06 by grifcj

PiotrZierhoffer, 2022-02-08:
Hi @grifcj 

I can confirm this behavior. This is caused by our move towards supporting both dotnet and Mono (still in progress).

If you're looking for an IDE I would suggest switching to vscode. This works fine on Linux and has reasonable debugging capabilities, with the biggest drawback of not being able to switch threads when on breakpoint.

We have not released the vscode configuration yet, but I can send it to you over email - nothing secret there, but requires some cleanup.

PiotrZierhoffer, 2022-02-09:
Actually I pushed the vscode configuraton on our master - only for Mono now, but we'll improve it with Windows support in future

grifcj, 2022-02-10:
@PiotrZierhoffer Works like a charm. Thanks for sharing. Tried it out on Ubuntu 20.04 / vscode 1.63.2.