renode/renode #293: AArch64 emulation support

Gerharddc, 2022-02-07:
From what I can tell, Renode does not yet support emulating AArach64 systems yet, only "classic" ARM. Is this correct and if so, are there any plans to support this in the future?

Along those same lines I am also curious about support for multiple cores.

PiotrZierhoffer, 2022-02-08:
Hi @Gerharddc , that's correct - AArch64 is not currently supported. It's definitely on the list, but I cannot give you any precise dates.

Multicore support is currently enabled in RISC-V. It will probably be added ARM cores along with aarch64.

ImmortalBuka, 2022-02-22:
@PiotrZierhoffer is multicore support enabled only for riscv64?

PiotrZierhoffer, 2022-02-22:
@ImmortalBuka at the moment this is correct, in the sense that we support RISC-V atomic instructions across cores. However, it's also possible to have separate cores with external synchronization mechanisms, modeled as bus peripherals. 

Sadly, at the moment we do not have public demos with this functionality.

bitbytedog, 2022-10-03:
Any update on aarch64 support? It would be very useful for my current work with the ZyncMP SoC.

PiotrZierhoffer, 2023-02-03:
Related to #431, we'll continue there as we're getting close to having an initial support released

closed 2023-02-03 by PiotrZierhoffer