renode/renode #295: Is it possible to insert in renode a verilator model of risc-v processor

mohamedElbouazzati, 2022-02-10:
Hello renode community,

I'm working on risc-v softcore with systemverilog and using a verilator simulator. 
I want to integrate my work to renode for further experiments in network side. 
I see in renode there are many existing platforms, and possibility to add some verilated peripherals such as uar but is it possible to integrate/build verilated CPU?
The platform is CV32E40P RISC-V CPU and  behavioral block written in system verilog to generate logfile that brings data from the CPU

Any helps, thanks a lot

PiotrZierhoffer, 2022-02-16:
Hi @mohamedElbouazzati 

Internally we do have a verilated Ibex working in Renode. This is not released yet, and will have to take some time to get released. It's a significant change in the internal infrastructure.

Regarding  additional blocks, we support custom verilated instructions implementations (in the context of CFUs, Custom Function Units), but I'm not sure if that scenario covers your needs.

tcal-x, 2022-02-16:
FYI @alanvgreen

mohamedElbouazzati, 2022-02-17:
@PiotrZierhoffer Many thanks for this information.

closed 2022-02-17 by mohamedElbouazzati