renode/renode #297: Add Word/DWord/QWord SPI Connections

xobs, 2022-02-13:
I have a SPI device that transmits 16 bits of data per transaction.

I keep track of which byte is getting sent, and decompose the two bytes of data into two transactions.

Due to differences in emulation speed, sometimes the SPIConnection adds a NULL byte. This throws off the connection and causes sync errors.

My solution is to extend `ISPIPeripheral` to include variants with different sizes:

```cs
namespace Antmicro.Renode.Peripherals.SPI
{
    public interface ISPIPeripheral : IPeripheral
    {
        byte Transmit(byte data);
        void FinishTransmission();
    }
    public interface IWSPIPeripheral : IPeripheral
    {
        ushort Transmit(ushort data);
        void FinishTransmission();
    }
    public interface IDSPIPeripheral : IPeripheral
    {
        uint Transmit(uint data);
        void FinishTransmission();
    }
    public interface IQSPIPeripheral : IPeripheral
    {
        ulong Transmit(ulong data);
        void FinishTransmission();
    }
}
```

Then I create a variant of `SPIConnector` that operates on `IWSPIPeripheral` to operate on words rather than bytes.

Unfortunately I need to patch Renode in order to do this, because Interfaces cannot be added via `include`, which means the Connection cannot work without addit it to Renode core.

Is there a better way to do this?

PiotrZierhoffer, 2022-02-13:
I can't test it at the moment, but patching SHOULD be solvable if you add `IInterestingType` to your interface. Then you should be able to just `include` it as well. 

This interface has nothing in it, it's just a flag, really.

The interface width is of course another topic we'd need to discuss internally. It kinda doesn't make sense, as it's very application specific, but of course it stems from the shape of the interface and our synchronization routine, so I understand there's a need for more flexibility.

PiotrZierhoffer, 2022-02-14:
Can you please verify your Renode version? Is this a relatively up-to-date build?

We worked on cross-file relationships in ad-hoc compiler, so this scenario should work as intended

xobs, 2022-02-14:
Renode tells me that it's `Renode, version 1.12.0.16906 (f41dea21-202202121922)`

PiotrZierhoffer, 2022-02-16:
Can you please try, after including the file with interfaces, to run:

`(monitor) EnsureTypeIsLoaded "Antmicro.Renode.Peripherals.SPI.IQSPIPeripheral"`

and only then include a file using it?

This is behavior that we might want to change, but it should help you with the issue.

xobs, 2022-02-18:
The `EnsureTypeIsLoaded` feature does work, and we now have a16-bit SPI bus that's functional, and is split into three different files: https://github.com/betrusted-io/xous-core/commit/9c1255bdd7896b857da70726657cbd6a051ff8b0

Thanks for the help!

closed 2022-02-18 by xobs