renode/renode #298: Testing framework function "WaitForLineOnUart": How to obtain first regex match?

rtzu6, 2022-02-14:
I want to use Renode and the Robot Testing Framework to test my software. To that end, I want to verify that a string matching a regex has been received. I also want to further use this string in the Robot Framework test. I started from the `QuarkC1000.robot` platform test, which suggest to use
```
${p}=    Wait For Line    <regex>    treatAsRegex=true
Log to Console    ${p.groups[0]}
```
This, however, does not really return `p.groups[0]` but `p.groups[1]`, so the second match which is no longer the string that I want.
The relevant code line is in `TerminalTester.cs`, in the `GetMatchGroups` function, where it skips the first result with the comment
`group '0' is the whole match; we return it in a separate field, so we don't want it here`, but I did not find the field that returns `group[0]`.

So how can I get the actual first match (`group[0]`) of a regex in WaitForLineOnUart or a similar function? What did I miss?

mateusz-holenko, 2022-02-22:
Hi @rtzu6, thanks for opening the issue!

If you want to access the whole line, use `p.line` as in:

```
${p}=    Wait For Line    <regex>    treatAsRegex=true
Log to Console    ${p.line}
```

PiotrZierhoffer, 2022-04-05:
Closing the issue. @rtzu6 if you have any other problems, please reopen or file a new one

closed 2022-04-05 by PiotrZierhoffer