renode/renode #299: GUI Support

LeoLee0097, 2022-02-19:
#Feature request GUI
Dear all
I am a freshman here. Could anyone tell me whether there is any GUI interface for Renode (just like the one in Cooja or the methods in NS3), or is there any plan for it?

PiotrZierhoffer, 2022-02-22:
Hi @LeoLee0097 . 

At the moment we are not looking into general purpose UIs, we have explored this topic in the past and it's quite difficult to come up with a one-size-fits-all solution: Renode handles many different scenarios, from single-node MCU simulation to complex, wired and wireless networks of heterogeneous systems, so we're focusing on other things.
It might change in the future.

LeoLee0097, 2022-02-26:
Dear @PiotrZierhoffer 
Thank you for your notifications; your answer helps me most because you comment on the current situation with fascinating features and reveal the future of ReNode.
I believe that all the assignments above will be well established and have the GUI one day with your great efforts. Come on!