renode/renode #300: Please inquire about UART communication related to STM32f4.

listenshin, 2022-02-20:
Hi!

I am learning Renode for the first time.

I am trying to output "Hello" through UART of STM32f4 to Renode.

The build uses CubeIDE, and the board is in progress by selecting STM32F407VGTX.

However, when starting the simulation, an error occurs as shown below.
(I checked the "Hello" output using libopencm3, but I am curious if the "Hello" output is not possible when using the CubeIDE build HAL Driver.)

[image]

Could you please tell me how to solve it?

In addition, we checked an example of Tensorflow using Zephyr based on STM32f4. I wonder if it is possible to operate Tensorflow using a different OS using CubeIDE rather than Zephyr.

thank you.

mateusz-holenko, 2022-02-22:
Hi @listenshin, thanks for reporting the issue!

First of all, warnings about uninmplemented registers are not necessary errors. It's typical for a model in Renode not to cover the whole logic of a device, but this doesn't mean the emulation is not usable.

Since I don't see any access to the UART device itself
It's possible that software is polling for some specific value when initializing HW before writing anything on UART.

Could you add the following commands to your script, run the simulation again and post logs:

```
(monitor) cpu LogFunctionNames true
(monitor) sysbus LogAllPeripheralsAccess true
```

The above enables dump of executed symbols as well as information about all read/writes that happen on the system bus.

listenshin, 2022-02-27:
Hi @mateusz-holenko , thanks for the reply.

I proceeded as instructed. Is the way I did it correct?
[image]

Here are the relevant logs. Can you tell by this log?
[image]
[image]
[image]
[image]

If the method is wrong, can you tell me how? Let me show you the log again.

And I wonder if it is possible to test with renode using the ELF file built with CubeIDE.
(My plan is to develop using renode with an ELF file built using HAL Driver with CubeIDE.)

thank you.

klei22, 2022-07-21:
Seeing this same issue when utilizing the CubeIDE, really looking forward to incorporating renode into my team's workflow.

Sharing repo which should be importable to STM32CUBEIDE:

https://github.com/klei22/stm32f4-renode-test

Found that the error is triggered from the following lines (157-160) in the main.c file:

```c
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
```

link to lines:
https://github.com/klei22/stm32f4-renode-test/blob/2309968c33265d26fae73ba1772bb79ca187fe62/Core/Src/main.c#L157-L160

Suggesting it would have something to do with the RCC_CLockConfig function, was hoping at this point to ask for further pointers?

Looking forward to being able to being able to use Renode for development, testing and debugging : )

KamilGalik, 2022-10-21:
Fix for this issue has been released in https://github.com/renode/renode/commit/64b348005e15194ee15156e934b8d5c9d65d15e2. Our automatic test shows that CubeMX projects with default config work well in Renode. @klei22 @listenshin could you check if it works well for you too?

klei22, 2022-10-25:
After using the latest, it appears that the particular board had the same error, at the very least looks like the uart output isn't coming through.  Saw some errors that seem to indicate the RCC wasn't defined, and after checking the platform file, seemed to be true...

However, tried again for another part number, namely the STM32F103C4TX, and it worked first try for an elf generated by the STM32CubeIDE. :tada: 

Was able to go through with robot tests as well. :robot: 

So while I'm still not able to print uart from the discovery example (and wondering if I just need to fill out the RCC portion of the platform file associated with the HAL_RCC_ClockConfig related errors), it seems that indeed CubeMX projects are functional now on my machine with the latest release.

fspagnoletti, 2022-12-10:
Hi everyone,
I'm having exactly the same problem trying to send a "Hello" message over uart and I am building the code with STM32CubeIDE.
STM32L4R9ZIJ6 is the platfoarm that I'm using and this is the first time facing Renode.
I leave below the screenshot of the error. 
Any hint is really appreciated.

[image]

MarasMac, 2023-03-28:
The value of "FLASH:ACR" depends on configured clock frequency. It means flash wait states which you can find in reference manual in "embedded flash memory interface" section. So you need to change the "FLASH:ACR" value in .repl script (eg. stm32f4.repl) to correct one. @fspagnoletti in your case it will be 0x0 intead of 0x5. The easiest way to fix it is check last two warnings related to "FLASH:ACR", one is write the other one is read and set the value which is written by WriteDoubleWord.

KamilGalik, 2023-07-11:
https://github.com/renode/renode-infrastructure/commit/dcecb01820ad8fd92bd5863b624b8b76423fdb6d adds a basic support for the latency handling to the STM32F4 flash controller model, so there is no more need for hardcoding the value in repl. Please update your platform and include the model in the repl file as in: https://github.com/renode/renode/blob/master/platforms/cpus/stm32f4.repl#L10-L12

dbarwacz, 2024-07-18:
I think I'm seeing the same problem but with bluepill (STM32f103). 

In debugger it reaches `Error_Handler` in `SystemClock_Config`. 
I tried https://github.com/renode/renode/issues/300#issuecomment-1630387799 by taking the value from 
`Write of value 0x30 to an unimplemented register FLASH:ACR (0x40022000) generated from SVD`
and adding it to resc config as `sysbus Tag <0x40022000, 0x40022003> "FLASH_ACR" 0x30` but this does not seem to work. What am I missing?

PotatoGuo, 2025-04-25:
@klei22 Thank you, solved my problem. I changed your test project to makefile and worked