renode/renode #301: Add udhcps to Versatile image

xobs, 2022-02-21:
As far as I can tell, there currently is no way to set up a DHCP server in Renode. Issue https://github.com/renode/renode/issues/148 is outstanding.

A simple workaround would be to create an image such as the Versatile image and run `udhcps`. This would provide a network server and a basic method to create a self-contained network within Renode.

PiotrZierhoffer, 2022-02-22:
Our Versatile image is quite dated, and not really easy to reproduce, but we will look into providing something on Zynq.

xobs, 2022-02-22:
I managed to build `busybox` with `libmusl` and a statically-linked binary. This image is the only file on the disk, and was installed with `make install`. I then created an `rcS` that does everything I need.

1. Get musl from https://musl.cc/ (https://musl.cc/armv5l-linux-musleabi-cross.tgz)
2. Get busybox from https://busybox.net/downloads/busybox-1.35.0.tar.bz2
3. Extract it
4. Build the default config with `CROSS_COMPILE=armv5l-linux-musleabi-`
5. Install with `make install`
6. Copy the `rcS` from below as `_install/etc/init.d/rcS` and mark it as `+x`
7. Make the image with `mkfs.jffs2 --pad=4194304 -n -d _install/ -o new-rootfs.jffs2`
8. Remove the `noinitrd` from the kernel command line

This results in an image with a complete `busybox` binary that's very small -- under 1MB. This comes with many networking tools including netcat and udhcpd.

rcS:

```sh
#!/bin/sh

mount -t proc none /proc
mount -t sysfs none /sys
mount -t devpts none /dev/pts

mdev -s -d

ifconfig eth0 192.168.0.1

cat <<!

Boot took $(cut -d' ' -f1 /proc/uptime) seconds

______      _                  _           _
| ___ \\    | |                | |         | |
| |_/ / ___| |_ _ __ _   _ ___| |_ ___  __| |
| ___ \\/ _ \\ __| '__| | | / __| __/ _ \\/ _\` |
| |_/ /  __/ |_| |  | |_| \\__ \\ ||  __/ (_| |
\\____/ \\___|\\__|_|   \\__,_|___/\\__\\___|\\__,_|

Dumb Little Betrusted Server

!

udhcpd -f &

# See https://busybox.net/FAQ.html#job_control for more
# information on this abomination
exec setsid sh -c 'exec sh </dev/tty1 >/dev/tty1 2>&1'
```