renode/renode #303: Silabs efm32gg missing peripherals, how to add?

tobermory, 2022-02-24:
I am a total newbie to renode. I have a .bin file built for a SiliconLabs efm32gg MCU.  There is a close match with

`
efm32gg942.repl
`
I built such a machine and load the binary and start.  My renode log window's first error is

`
20:36:45.5586 [WARNING] sysbus: [cpu: 0x2D4] (tag: 'MSC') WriteDoubleWord to non existing peripheral at 0x400C0040, value 0x1.
`
The .repl file has a mention of the 'tag' MSC:

`
sysbus:
    init:
        Tag <0x400c0000,0x400c03ff> "MSC"
`

Does this mean that this peripheral is known to occupy that part of memory (peripherals are all memory-mapped on this board), and has a tag (label?) MSC, but there is no actual peripheral in renode to service reads and writes to that region?

How do I go about implementing such a peripheral?

PiotrZierhoffer, 2022-02-24:
Hi @tobermory .

Your analysis is exactly correct. These kind of warnings are very much expected in Renode - they indicate that your software tried to access a block that we do not support. They are expected, because we rarely implement full SoCs - just the parts that are interesting from the perspective of the software we're running.

In this case, I believe, we're talking about memory controller - we most likely never needed it. Even if it's accessed, many peripherals are just configured and left alone, so they can stay as tags or simple Python peripherals, mocking some logic.

If you actually need to talk to this block, you can implement it in C# (this is the preferred way) or in Python. We also have a very experimental interface to Rust: https://github.com/antmicro/renode-rust-example

A somewhat high-level description of the process is described here: https://renode.readthedocs.io/en/latest/advanced/writing-peripherals.html

You can also refer to the Betrusted docs: https://betrusted.io/xous-book/ch04-04-writing-cs-peripherals.html - we are working on incorporating this chapter in Renode documentation as well.

tobermory, 2022-02-24:
Thanks for the prompt reply. The lack of this peripheral, and others, has forced me to ask a useful question: can my code run just as well without these peripherals? Most of my app's logic is just to serial I/O to various (oceanographic) devices.  Any use of e.g. the MSC (yes, memory controller, as you said, it's the peripheral that erases and writes pages of flash) is solely in routines provided by the vendor (in my case Silabs) to initialize the chip, my own app logic makes no use of the MSC itself.

So, if writes are no-ops and reads always give 0, can my application 'work' on Renode, even with some peripherals missing?  A good question to think about.

Thanks again

tobermory, 2022-02-27:
OK, taking a lead from Issue 284, I forked renode, and added a .cs file for a peripheral I needed in the Infrastructure directory tree.  I am unsure how to then build my code, and have renode recognize it if I name that peripheral in a .repl file.  If I just do 'build.sh' from the root dir again, it succeeds, yet i get the feeling it hasn't seen my new .cs file.  In fact if I put a garbage .cs file in same directory, that cannot possibly be a valid .cs file, I still get no build errors.

Any help on adding peripherals, and especially building them, appreciated.

PiotrZierhoffer, 2022-02-27:
Please find the Peripherals.csproj file and list it there.

But the fun thing about Renode is that you can build it once and then load your cs file from the Monitor with `include @path/to/your/file.cs`. It will get automatically compiled. The you'll be able to use it in repl files etc.

Keep in mind you have to restart Renode to load a new version of the same file (unless the compilation fails).

tobermory, 2022-02-27:
Sorry, I am not following.  What is the utility of the 'include' directive? I just want to be able to define my peripheral in the .repl file.

PiotrZierhoffer, 2022-02-27:
You can either compile your model with build.sh (then it needs to be listed in Peripherals.csproj) or you can compile it in Renode runtime. Run Renode, in the console that pops up `include` your model, then you can use it as any other model in the framework.