renode/renode #307: Cortex-M0+ sometimes skips `SVC` instruction

xobs, 2022-03-01:
Working with an embedded operating system, it makes syscalls by using the `SVC` instruction. This should raise an interrupt which should be handled by handler `0xb`.

This operating system checks `$xpsr` to determine whether it should make a syscall by issuing a `SVC` instruction or call the function correctly.

It is able to make a syscall once, however subsequent syscalls cannot be made. This works on real hardware, which leads me to believe there's something going wrong in the unmasking of exceptions when returning from the handler:

```
(gdb) disassemble $pc-2,+6
Dump of assembler code from 0x16858 to 0x1685e:
   0x00016858 <OS_SYSTEMCALL_ACTIVATETASK+22>:  bne.n   0x16866 <OS_SYSTEMCALL_ACTIVATETASK+36>
=> 0x0001685a <OS_SYSTEMCALL_ACTIVATETASK+24>:  svc     7
   0x0001685c <OS_SYSTEMCALL_ACTIVATETASK+26>:  adds    r2, r0, #0
End of assembler dump.
(gdb) p/x $xpsr
$70 = 0x61000000
(gdb) p/x $primask
$71 = 0x0
(gdb) stepi
141     in System/Os/rtos/src/os_syscall.h
(gdb) disassemble $pc-2,+6
Dump of assembler code from 0x1685a to 0x16860:
   0x0001685a <OS_SYSTEMCALL_ACTIVATETASK+24>:  svc     7
=> 0x0001685c <OS_SYSTEMCALL_ACTIVATETASK+26>:  adds    r2, r0, #0
   0x0001685e <OS_SYSTEMCALL_ACTIVATETASK+28>:  movs    r3, #15
End of assembler dump.
(gdb)
```

In the above snippet, you can see there's a `svc 7` located at offset `0x1685a`. The `$xpsr` bits are 0, indicating it's not currently executing an interrupt. `$primask` is 0 indicating that interrupts are not masked.

GDB is instructed to step exactly one instruction (`stepi`), which ought to step into the `SVC_Handler` routine. However it instead simply skips over the instruction as if it were masked, or as if it were a noop.

tobermory, 2022-03-01:
I am new to renode and am also experimenting with RTOS-using applications.  I do not have an answer for you, but am curious to know whether the problem goes away if you move to a CM3, emulated of course.

PiotrZierhoffer, 2022-03-02:
Hey @xobs , as we discussed today, there is an option `cpu ShouldEnterDebugMode`, but it's **not** set by default for Cortex-M. So this is most likely some other issue, and we'll look into this.

@tobermory highly unlikely, the codebase is almost the same, with some differences mostly in enabled features.

closed 2022-03-18 by RenodeBot

ajelinski, 2022-03-18:
I've just merged changes which fix a problem with the `PRIMASK` register.
With these changes, GDB now properly shows that `PRIMASK` is set when the `svc 7` is processed.

With `PRIMASK` unset, after the `svc 7` instruction there's a proper jump to the interrupt handler.
For testing purposes, it can be now properly unset with `set $primask=0x0` through GDB or `cpu PRIMASK 0x0` through Renode Monitor.

Thanks a lot for reporting this issue @xobs.

commit 39e2fb71 (2022-03-18): [GH#307] Arm-M: Use the tlib's `PRIMASK` register

Fixes #307.