renode/renode #309: About metrics analyzer struct.error

dokodokoR, 2022-03-09:
I tryed to use renode and metrics analyzer.
But struct error occured like below.  
How do I activate the Metrics Analyzer?

```bash
// renode 
$ renode
(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/boards/stm32f4_discovery-kit.repl
(machine-0) sysbus LoadELF @renode-example.elf
(machine-0) machine EnableProfiler dump_file
(machine-0) start
(machine-0) quit // about 1s later
```

```bash
// bash
$ python3 tools/metrics_analyzer/metrics_visualizer/metrics-visualizer.py dump_file
Traceback (most recent call last):
  File "tools/metrics_analyzer/metrics_visualizer/metrics-visualizer.py", line 42, in <module>
    generate_report(metricsParser, options)
  File "tools/metrics_analyzer/metrics_visualizer/metrics-visualizer.py", line 34, in generate_report
    show_executed_instructions(metricsParser, options, onePlotFigureSize, fontSize)
  File "/Users/masaya/lab/renode/tools/metrics_analyzer/metrics_visualizer/assets/instructions.py", line 10, in show_executed_instructions
    cpus, instructionEntries = metricsParser.get_instructions_entries()
  File "tools/metrics_analyzer/metrics_parser/__init__.py", line 17, in get_instructions_entries
    cpus, _ = self._parseHeader(f)
  File "tools/metrics_analyzer/metrics_parser/__init__.py", line 63, in _parseHeader
    cpus[cpuId] = self._read('{}s'.format(cpuNameLength), f)[0].decode()
  File "tools/metrics_analyzer/metrics_parser/__init__.py", line 83, in _read
    return unpack(format, file.read(calcsize(format)))
struct.error: unpack requires a buffer of 729116771 bytes
```

Thanks

ref. 

- https://interrupt.memfault.com/blog/intro-to-renode
- https://renode.readthedocs.io/en/latest/basic/metrics.html

PiotrZierhoffer, 2022-03-12:
Hi @dokodokoR 

Can you please paste your Renode version number? You can run the `version` command from the Monitor to print it out.

dokodokoR, 2022-03-12:
@PiotrZierhoffer 

renode version is here
```
(machine-0) version
Renode, version 1.12.0.41386 (44d6786a-202104022259)
```

dokodokoR, 2022-03-14:
@PiotrZierhoffer 

 I ran the above command again and it worked.

Sorry, I could not figure out the cause.

mateusz-holenko, 2022-03-29:
I guess we can close this one.

closed 2022-03-29 by mateusz-holenko