renode/renode #310: strange startup behavior

EduardShaid, 2022-03-11:
Someone please help!

Installed renode.
Then I give the command in the terminal:
**(monitor) start @scripts/single-node/stm32f4_discovery.resc**

but the response is an error:

**12:43:53.5456 [INFO] System bus created.
12:43:54.3719 [INFO] Downloading https://dl.antmicro.com/projects/renode/svd/STM32F40x.svd.gz.
12:43:54.4983 [ERROR] Failed to download from https://dl.antmicro.com/projects/renode/svd/STM32F40x.svd.gz, reason: SecureChannelFailure (attempt 1/5)
12:43:54.4983 [INFO] Downloading https://dl.antmicro.com/projects/renode/svd/STM32F40x.svd.gz.**

PiotrZierhoffer, 2022-03-12:
I believe you're running on Windows, using the 1.12 package? 

Please try getting the latest nightly MSI from https://builds.renode.io, or build from sources - this should solve your issue.

edbek, 2022-03-14:
Yes, that's right, I'm using Windows and 1.12 package.
Thanks a lot for the advice!

EduardShaid, 2022-03-14:
I'm trying to build a project in Windows 10 from sources.
But an error occurs when building the project: the .NET 4.5 is outdated and is no longer supported.

mateusz-holenko, 2022-04-05:
Could you share a detailed log from your build, @EduardShaid? I tired to reproduce this (although on Windows 11) and didn't notice any warnings/errors related to .NET 4.5 being outdated and not supported.

mateusz-holenko, 2022-04-26:
@EduardShaid Did you follow steps described in our documentation:https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html (please not there is a tab describing the procedure for Windows)?

We've recently verified that this works both on Windows 10 and Windows 11.

edbek, 2022-04-26:
Colleagues, thank you very much for your help ! This topic can be closed.

closed 2022-04-26 by mateusz-holenko

breakersun, 2023-07-03:
still happens at the latest release v1.9.0

https://github.com/renode/renode/releases/download/v1.9.0/renode_1.9.0.msi

PiotrZierhoffer, 2023-07-31:
Hi @breakersun 

1.9 is over 3 years old. The current version is 1.13 and in this or the next week we should release 1.14. 

You can either build from source or use nightly builds from https://builds.renode.io