renode/renode #311: Upgrade to RobotFramework 4 causes Error

Xcross74, 2022-03-23:
Hi All,

I have been running Renode tests on RobotFramework v3.1 but when upgrading to v4.1.3 I get an error immediately and don't know why.
[image]
If I revert back to v3 I can run my tests again. I'm new to this framework so maybe someone can point me in the right direction ?

Below is a pip freeze of installed packages, perhaps one of them is incompatible ? Thanks !
[python_installation.txt](https://github.com/renode/renode/files/8333353/python_installation.txt)

PiotrZierhoffer, 2022-03-23:
Hi @Xcross74 

We have changed the supported Robot version some time ago, from 3.X to 4.0.1. The API changed significantly and we have no backwards compatibility

It happened last year - which version of Renode are you using? I would assume it's either the official release or some older commit?

Please take a look at https://github.com/renode/renode/blob/master/tests/requirements.txt for current pip requirements.

Xcross74, 2022-03-23:
thank you for the reply.
I've double checked the requirements and just in case did a pip install -r requirements.txt with the ones from your link, however still no difference.
Attached is the Renode version I am using. It is the v1.12. Is this one not compatible with Robot v4 do you think ?
[image]

PiotrZierhoffer, 2022-03-29:
This is our official release, which relies on Robot 3.1.

I'd recommend using one of your latest builds from https://builds.renode.io/

The Renode 1.13 is going to happen quite soon, and it will rely on Robot 4.0.1

closed 2022-04-06 by Xcross74