renode/renode #312: Memory peripheral missing in MIMXR1064-EVK

GaelRap, 2022-03-29:
I'm trying to emulate a simple hello_world code on the board MIMXRT1064-EVK which should write on lpuart4, but it seems that peripherals are not defined. I use the script `start_imx.resc` in the monitor as `start @start_imx.resc` which is compose by :

```
# Filename: start_imx.resc
$bin?=@/tmp/zephyrproject/hello_world_mimxrt1064_evk.elf

# Create Machine & Load config
using sysbus
mach create
machine LoadPlatformDescription @platforms/cpus/imxrt1064.repl
sysbus LoadELF $bin
cpu VectorTableOffset `sysbus GetSymbolAddress "_vector_table"`

# Create a terminal window showing the output of LPUART4
showAnalyzer sysbus.lpuart4

macro reset
"""
    sysbus LoadELF $bin
"""

runMacro $reset
```

 I get the following `warnings` and `error`

```
12:18:03.2610 [INFO] System bus created.
12:18:04.0438 [INFO] sysbus: Loading segment of 22604 bytes length at 0x70000000.
12:18:04.0454 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x70000000, 0x7000584B>.
12:18:04.0458 [INFO] sysbus: Loading segment of 48 bytes length at 0x7000584C.
12:18:04.0459 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x7000584C, 0x7000587B>.
12:18:04.0460 [INFO] sysbus: Loading segment of 3920 bytes length at 0x80000030.
12:18:04.0461 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x80000030, 0x80000F7F>.
12:18:04.1832 [WARNING] cpu: Tried to set VTOR address at 0x70002000 which does not lay in memory. Aborted.
12:18:04.2755 [INFO] sysbus: Loading segment of 22604 bytes length at 0x70000000.
12:18:04.2756 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x70000000, 0x7000584B>.
12:18:04.2756 [INFO] sysbus: Loading segment of 48 bytes length at 0x7000584C.
12:18:04.2757 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x7000584C, 0x7000587B>.
12:18:04.2757 [INFO] sysbus: Loading segment of 3920 bytes length at 0x80000030.
12:18:04.2758 [WARNING] sysbus: Tried to access bytes at non-existing peripheral in range <0x80000030, 0x80000F7F>.
12:18:04.2903 [ERROR] cpu: PC does not lay in memory or PC and SP are equal to zero. CPU was halted.
12:18:04.2903 [INFO] cpu: Setting initial values: PC = 0x0, SP = 0x0.
12:18:04.2906 [WARNING] cpu: Patching PC 0x0 for Thumb mode.
12:18:04.2925 [INFO] machine-0: Machine started.
```

Do you have any clue ?

mateusz-holenko, 2022-04-05:
Hi @GaelRap, thanks for reporting the issue!

I believe that loading the [mimxrt1064_evk.repl](https://github.com/renode/renode/blob/master/platforms/boards/mimxrt1064_evk.repl) platform (that defines board-specific memory regions) instead of mimxrt1064.repl core should resolve your issue:
```
machine LoadPlatformDescription @platforms/boards/imxrt1064_evk.repl
```

xiaomingzhuang, 2022-04-18:
Hi @mateusz-holenko, I have used mimxxrt1064_evk.repl, the warnings of "try to  access bytes at non-existing peripheral" disappeared , but still an error as @GaelRap still there
12:18:04.2903 [ERROR] cpu: PC does not lay in memory or PC and SP are equal to zero. CPU was halted.
12:18:04.2903 [INFO] cpu: Setting initial values: PC = 0x0, SP = 0x0.
12:18:04.2906 [WARNING] cpu: Patching PC 0x0 for Thumb mode.
12:18:04.2925 [INFO] machine-0: Machine started.

Do you have any clue?

xiaomingzhuang, 2022-04-18:
Sorry,I got the answer from NXP community, we can check the "Link to RAM" option which can be found on the Linker -> Managed Linker Script page of project properties

PiotrZierhoffer, 2022-04-26:
I guess we can close this issue for now. Feel free to follow-up or reopen a new issue.

@GaelRap you are trying to run Zephyr's hello-world sample. Please take a look at https://zephyr-dashboard.renode.io/ where you can see it being tested in Renode and you can download all the necessary scripts and platform files.

closed 2022-04-26 by PiotrZierhoffer