renode/renode #313: Missing dependency when building for windows

ohunter, 2022-03-29:
I am trying to build the project to test if it would be useful for projects I'm involved with at work.
I am running Windows 10 Enterprise (19044.1586), using MSys2 as my bash terminal. I am also using the following commit `8e639071cd882b5ed8133d3863965599bbb0179c` which is master at the time of writing
After having fixed some issues in `build.sh` (when running `msbuild` from msys2 the `/p:` options need to be replaced with `//p:`)
After making these changes and I try to run `build.sh` I then encounter the issue at hand. 
```sh
C:\Users\olejeh\Downloads\Gits\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): warning : /usr/lib/gcc/x86_64-pc-msys/11.2.0/../../../x86_64-pc-msys/bin/ld: cannot find -lwinpthread
```
This would indicate that to build the system on windows, the `winpthread` project is needed yet it is not mentioned anywhere that I can see. I can also inform that it is not as just installing it from msys2, although that is for me to figure out why.

The complete build log can be seen [here](https://github.com/renode/renode/files/8371116/renode_build_log.txt)

ohunter, 2022-03-30:
So just to be sure I just tried with Cygwin as well. No luck. After installing winpthread for either environment and modifying `LIBRARY_PATH` before running `build.sh` so that winpthread could be located I get a quite a few undefined references. Not really sure what to do at this point.

PiotrZierhoffer, 2022-04-05:
Hi @ohunter .

I believe you are not using MinGW, is that correct? You will find that in the documentation (https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html) that we currently support building with MinGW, with win32 threading and sjlj exception handling.

If that doesn't help, please post the build log and we'll try to look into it.

ohunter, 2022-04-06:
Where in the documentation does it say that? the only reference I can find to anything windows specific is a sentence that states you only support building with Cygwin. Nonetheless, Msys2 uses MinGW libraries internally so that shouldn't really be an issue.

PiotrZierhoffer, 2022-04-26:
We'll try to add a feature to the documentation that will focus on specific tabs.

Please let me know if the instructions helped.

closed 2026-07-01 by PiotrZierhoffer