renode/renode #317: Ethernet Transmit Complete Status never set?

minifake, 2022-04-08:
When I debugging Ethernet, I found the TS(Transmit Status) bit in DMASR never set after a successful sent. Which cause program never go to TxCallback as expexted. 

I checked the renode-infrastructure code, this bit indeed never set in the code. It only sets TBUS bit as shown in the link below. 
https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Network/SynopsysEthernetMAC.cs#L398

Is that an issue?

PiotrZierhoffer, 2022-04-26:
Hi @minifake 

We'll see into adding that. Please be aware that this model is quite old and incomplete, so there might be other issues you encounter.

mateusz-holenko, 2022-05-06:
In https://github.com/renode/renode/commit/f8571962d2811a6e67f61f8724434022a3c05045 we added setting the `TransmitStatus` flag when sending frames in `SynopsysEthernetMAC`.

Could you verify @minifake if the current version works for you?

minifake, 2022-06-10:
Sorry for the late response. I've tested the newest version, it works! I will close this issue. Thanks a lot.

closed 2022-06-10 by minifake