renode/renode #319: .NET 4.5 is deprecated and is now unavailable

xobs, 2022-04-21:
Renode is built against .NET 4.5, which is no longer available for install. Furthermore, it's not even an option in Visual Studio 2022. It cannot be obtained from https://dotnet.microsoft.com/en-us/download/dotnet-framework/net45

A simple solution is to simply specify `4.6.2` as a dependency, which I have verified works and results in a functioning Renode.

For example:

```patch
diff --git a/src/Emulator/Cores/cores-arm-m.csproj b/src/Emulator/Cores/cores-arm-m.csproj
index d118ccf7..f65e82da 100644
--- a/src/Emulator/Cores/cores-arm-m.csproj
+++ b/src/Emulator/Cores/cores-arm-m.csproj
@@ -8,7 +8,7 @@
     <SchemaVersion>2.0</SchemaVersion>
     <OutputType>Library</OutputType>
     <AssemblyName>cores-arm-m</AssemblyName>
-    <TargetFrameworkVersion>v4.5</TargetFrameworkVersion>
+    <TargetFrameworkVersion>v4.6.2</TargetFrameworkVersion>
     <PropertiesLocation>..\..\..\..\..\output\properties.csproj</PropertiesLocation>
   </PropertyGroup>
   <Import Project="$(PropertiesLocation)" />
```

xobs, 2022-04-21:
According to https://dotnet.microsoft.com/en-us/platform/support/policy/dotnet-framework, apparently 4.6.2 will reach end-of-life *tomorrow*. Well that's not great.

I have successfully built using `v4.8`, which has no date set for the end of support.

xobs, 2022-04-25:
Thanks. I can install Visual Studio 2019, which has two more years of mainstream support left: https://docs.microsoft.com/en-us/lifecycle/products/visual-studio-2019

Visual Studio 2017 has reached end-of-life: https://docs.microsoft.com/en-us/lifecycle/products/visual-studio-2017

I misread the website, and `v4.6.2` is currently the oldest supported .NET available from Microsoft. `v4.6.1` went EOL last week.

PiotrZierhoffer, 2022-04-26:
We will see if `4.6.2` is supported by xbuild, but I guess I'll close this issue for now.

closed 2022-04-26 by PiotrZierhoffer