renode/renode #321: Use of runtime compiled peripherals as parameter

snooka2197, 2022-04-21:
Hi,
i am working on a STM32L0x1 platform file and created two peripherals STM32L0x1_RCC & STM32L0x1_RTC that are quite similar to the already supported STM32F4_RTC & STM32F4_RCC.
In the stm32f429.repl they are used in the following way (RCC peripheral is capable to enable/disable RTC peripheral):
```
rtc: Timers.STM32F4_RTC @ sysbus 0x40002800
    AlarmIRQ -> nvic@41

rcc: Miscellaneous.STM32F4_RCC @ sysbus 0x40023800
    rtcPeripheral: rtc
```

I have included my peripherals in the script:
i $ORIGIN\..\peripherals\STM32L0x1_RTC.cs
i $ORIGIN\..\peripherals\STM32L0x1_RCC.cs

and adapted the platform accordingly:
```
rtc: Timers.STM32L0x1_RTC @ sysbus <0x40002800, +0x400>
    AlarmIRQ -> nvic@41
	
rcc: Miscellaneous.STM32L0x1_RCC @ sysbus 0x40021000
    rtcPeripheral: rtc
```

However, running this will result in the following error message.
```
The type or namespace name `STM32L0x1_RTC' could not be found. Are you 
missing an assembly reference?
```

Without the "rtcPeripheral" parameter the simulation will run just fine and list all the included peripherals (loosing the RCC-disable feature).
What is the issue here? Is it not possible to use included peripherals as parameters in general?

mateusz-holenko, 2022-04-21:
Do I understand correctly that you are getting this error when trying to compile `STM32L0x1_RCC.cs`?

If so I would suggest either joining those two files together or adding the following command
```
EnsureTypeIsLoaded "Antmicro.Renode.Peripherals.Timers.STM32L0x1_RTC"
```
before the second `include` command.

snooka2197, 2022-04-21:
Hi,
yes, this was during compiling of the cs file.
Thanks, this solved the issue.
i missed the "EnsureTypeIsLoaded"-feature. Is it documented somewhere?

closed 2022-04-25 by snooka2197