renode/renode #322: stm32f429 [ERROR] cpu: CPU abort []: stl_phys address is not aligned

jhgorse, 2022-04-21:
Greetings,

I have code which executes on hardware which is struggling with GPIO init here on `renode`.

I have managed to get `arm-none-eabi-gdb` to work on the command line. It fails to capture the backtrace of the issue or location precisely because SIGABRT requires libc unwinding, which ARM did not bake into this gcc.

The partial `log 0` is here:
```
16:36:53.0057 [WARNING] sysbus: Read from an unimplemented register GPIOG:OSPEEDR (0x40021808), returning a value from SVD: 0x0.
16:36:53.0057 [WARNING] sysbus: Write of value 0x0 to an unimplemented register GPIOG:OSPEEDR (0x40021808) generated from SVD.
16:36:53.0058 [WARNING] sysbus: Read from an unimplemented register GPIOG:OSPEEDR (0x40021808), returning a value from SVD: 0x0.
16:36:53.0058 [WARNING] sysbus: Write of value 0x20000 to an unimplemented register GPIOG:OSPEEDR (0x40021808) generated from SVD.
16:36:53.0058 [WARNING] sysbus: Read from an unimplemented register GPIOG:OTYPER (0x40021804), returning a value from SVD: 0x0.
16:36:53.0058 [WARNING] sysbus: Write of value 0x0 to an unimplemented register GPIOG:OTYPER (0x40021804) generated from SVD.
16:36:53.0059 [WARNING] sysbus: Read from an unimplemented register GPIOG:OTYPER (0x40021804), returning a value from SVD: 0x0.
16:36:53.0059 [WARNING] sysbus: Write of value 0x0 to an unimplemented register GPIOG:OTYPER (0x40021804) generated from SVD.
16:36:53.0059 [WARNING] sysbus: Read from an unimplemented register GPIOG:PUPDR (0x4002180C), returning a value from SVD: 0x0.
16:36:53.0059 [WARNING] sysbus: Write of value 0x0 to an unimplemented register GPIOG:PUPDR (0x4002180C) generated from SVD.
16:36:53.0060 [WARNING] sysbus: Read from an unimplemented register GPIOG:PUPDR (0x4002180C), returning a value from SVD: 0x0.
16:36:53.0060 [WARNING] sysbus: Write of value 0x0 to an unimplemented register GPIOG:PUPDR (0x4002180C) generated from SVD.
16:36:53.0060 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0060 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0061 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0061 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0061 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0061 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0062 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0062 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0062 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0063 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0063 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0063 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0063 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021494
16:36:53.0063 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8020FE4
16:36:53.0064 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021002
16:36:53.0064 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021014
16:36:53.0064 [INFO] cpu: Entering function HAL_GPIO_Init at 0x802101E
16:36:53.0064 [INFO] cpu: Entering function HAL_GPIO_Init at 0x8021028
16:36:53.0065 [ERROR] cpu: CPU abort [PC=0x802102C]: stl_phys address is not aligned: 1000ff5f.
```

It may be fully reproduced from this repository:
https://github.com/jhgorse/renode_stm32f429_vscode

Resc file here:
https://github.com/jhgorse/renode_stm32f429_vscode/blob/main/stm32f429_atollic_9.3.0.resc

What is next?

Cheers,
Joe

mateusz-holenko, 2022-04-27:
Hi @jhgorse, thanks for reporting the issue.

In dfd17cf7556e3b2cd2b9f4aa14ac8ae89c20bcfb we add support for unalingned stl/ldl operations - your example should work now. Could you verify this?

jhgorse, 2022-04-27:
Verified. SysTick seems to fire. On to the next tests. Thanks. =)

closed 2022-04-27 by jhgorse