renode/renode #324: out of memory attempting to use Litex Framebuffer (640x480x32bit)

swetland, 2022-05-01:
In a configuration with 32MB ram, VexRiscv, etc (see attached scripts), running a tiny binary that's just spinning, attempting to bring up the video framebuffer by writing the hres/vres/dmabase/enable registers (the only bits that look to be wired up based on the source here ( ):
```
sysbus WriteDoubleWord 0xf000301c 0x280
sysbus WriteDoubleWord 0xf000303c 0x1e0
sysbus WriteDoubleWord 0xf000305c 0x40c00000
sysbus WriteDoubleWord 0xf0003018 1
```

I get this out of memory exception:
```
Insufficient memory to continue the execution of the program.
  at (wrapper alloc) System.Object.AllocVector(intptr,intptr)
  at Antmicro.Renode.Peripherals.Video.AutoRepaintingVideo.Reconfigure (System.Nullable`1[T] width, System.Nullable`1[T] height, System.Nullable`1[T] format, System.Boolean autoRepaint) [0x000bc] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Peripherals.Video.LiteX_Framebuffer.<DefineRegisters>b__11_4 (System.Boolean _, System.Boolean val) [0x000ff] in <678e702a5c3c419196cff60b5a5669de>:0 
  at Antmicro.Renode.Core.Structure.Registers.PeripheralRegister+RegisterField`1[T].CallWriteHandler (System.UInt32 oldValue, System.UInt32 newValue) [0x0002c] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Core.Structure.Registers.PeripheralRegister.WriteInner (System.Int64 offset, System.UInt32 value) [0x001df] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Core.Structure.Registers.DoubleWordRegister.Write (System.Int64 offset, System.UInt32 value) [0x00000] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Core.Structure.Registers.DoubleWordRegisterCollection.TryWrite (System.Int64 offset, System.UInt32 value) [0x00010] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Core.Structure.Registers.DoubleWordRegisterCollection.Write (System.Int64 offset, System.UInt32 value) [0x00000] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Peripherals.Video.LiteX_Framebuffer.WriteDoubleWord (System.Int64 address, System.UInt32 value) [0x00006] in <678e702a5c3c419196cff60b5a5669de>:0 
  at Antmicro.Renode.Peripherals.Bus.SystemBus.WriteDoubleWord (System.UInt64 address, System.UInt32 value) [0x000a6] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at (wrapper dynamic-method) System.Object.CallSite.Target(System.Runtime.CompilerServices.Closure,System.Runtime.CompilerServices.CallSite,object,object,object)
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction (System.Runtime.CompilerServices.CallSite& callsite, System.Type binderType, System.Int32 knownType, Dynamitey.Internal.Optimization.InvokeHelper+LazyBinder binder, Dynamitey.InvokeMemberName name, System.Boolean staticContext, System.Type context, System.String[] argNames, System.Object target, System.Object[] args) [0x000df] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite (System.Object target, Dynamitey.InvokeMemberName name, System.Object[] args, System.String[] tArgNames, System.Type tContext, System.Boolean tStaticContext, System.Runtime.CompilerServices.CallSite& callSite) [0x0004f] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Dynamitey.Dynamic.InvokeMemberAction (System.Object target, Dynamitey.String_OR_InvokeMemberName name, System.Object[] args) [0x00018] in <1d0e1ef63ad2470a9ccec81105ee1302>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext (Dynamitey.InvokeContext context, System.Reflection.MethodInfo method, System.Object[] parameters) [0x00023] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.InvokeMethod (System.String name, System.Reflection.MethodInfo method, System.Collections.Generic.List`1[T] parameters) [0x0001a] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction (System.Type type, System.String name, System.Collections.Generic.IEnumerable`1[T] p) [0x00216] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction (System.Type device, System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00023] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName (System.String name, System.Collections.Generic.IEnumerable`1[T] p, AntShell.Commands.ICommandInteraction writer) [0x00060] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x00165] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001df] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x0019d] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at Antmicro.Renode.UserInterface.Monitor.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ci) [0x00000] in <ef089708f06a4b288ece4adcf3edaa2a>:0 
  at AntShell.Shell.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ic) [0x000bd] in <9a3ac418955d4a88b7929b87a631c91f>:0 
  at AntShell.CommandLine.HandleControlSequence (AntShell.Helpers.ControlSequence seq) [0x00a45] in <9a3ac418955d4a88b7929b87a631c91f>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.HandleInput (System.Object input) [0x0002e] in <9a3ac418955d4a88b7929b87a631c91f>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.Run (System.Boolean stopOnError) [0x00017] in <9a3ac418955d4a88b7929b87a631c91f>:0 
  at AntShell.Shell.Start (System.Boolean stopOnError) [0x00102] in <9a3ac418955d4a88b7929b87a631c91f>:0 
  at Antmicro.Renode.UI.CommandLineInterface+<>c__DisplayClass0_1.<Run>b__3 (System.Object x) [0x00000] in <4650942841e04c76931da7700721cda6>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00025] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ThreadHelper.ThreadStart (System.Object obj) [0x0000f] in <aeea2e33ea654cfabae63b60ba0e127a>:0
```

swetland, 2022-05-01:
script.resc
```
using sysbus

mach create
machine LoadPlatformDescription @x/cpu.resc

sysbus SilenceRange <0xf0004000 0x100> # SDRAM
sysbus SilenceRange <0xf0009000 0x100> # ETH PHY

showAnalyzer uart

$bios=@/work/os-workshop/out/bios.bin

macro reset
"""
    sysbus LoadBinary $bios 0x40000000
    cpu PC 0x40000000
"""

runMacro $reset
```

cpu.resc
```
rom: Memory.MappedMemory @ sysbus 0x0
    size: 0x00008000

sram: Memory.MappedMemory @ sysbus 0x10000000
    size: 0x00008000

ram: Memory.MappedMemory @ sysbus 0x40000000
    size: 0x02000000

emulator_ram: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x00004000

cpu: CPU.VexRiscv @ sysbus
    timeProvider: cpu_timer
    cpuType: "rv32ima"
    privilegeArchitecture: PrivilegeArchitecture.Priv1_10

timer0: Timers.LiteX_Timer @ sysbus 0xF0002000
    frequency: 50000000
    -> cpu@1

uart: UART.LiteX_UART @ sysbus 0xF0002800
    -> cpu@0

cpu_timer: Timers.LiteX_CPUTimer @ sysbus 0xf0000800
    frequency: 75000000
    // IRQ #100 is Machine Timer Interrupt
    -> cpu@100

spi: SPI.LiteX_SPI @ sysbus 0xf0008000

litex_video: Video.LiteX_Framebuffer @ sysbus 0xf0003000
    format: PixelFormat.XBGR8888
    memory: ram
```

swetland, 2022-05-01:
Using a smaller size (256x192), I don't get out of memory issues, but things blow up when I try to launch the display:
```
sysbus WriteDoubleWord 0xf000301c 256
sysbus WriteDoubleWord 0xf000303c 192
sysbus WriteDoubleWord 0xf000305c 0x40c00000
sysbus WriteDoubleWord 0xf0003018 1

showAnalyzer sysbus.litex_video
```

```
Value cannot be null.
Parameter name: native
  at (wrapper managed-to-native) System.Runtime.InteropServices.Marshal.copy_from_unmanaged_fixed(intptr,int,System.Array,int,void*)
  at System.Runtime.InteropServices.Marshal.copy_from_unmanaged (System.IntPtr source, System.Int32 startIndex, System.Array destination, System.Int32 length) [0x00000] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Runtime.InteropServices.Marshal.Copy (System.IntPtr source, System.Byte[] destination, System.Int32 startIndex, System.Int32 length) [0x00009] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at Cairo.ImageSurface.get_Data () [0x00017] in <b1a7b9dffbf2400c95bfd132d459b03b>:0 
  at Xwt.GtkBackend.ImageBuilderBackend.CreatePixbuf (Cairo.ImageSurface sf) [0x00000] in <4e388ed9219c471b9cbcf688a60afe69>:0 
  at Xwt.GtkBackend.GtkImage.RenderFrame (Xwt.Backends.ApplicationContext actx, System.Double scaleFactor, System.Double width, System.Double height) [0x00077] in <4e388ed9219c471b9cbcf688a60afe69>:0 
  at Xwt.GtkBackend.GtkImage.GetBestFrame (Xwt.Backends.ApplicationContext actx, System.Double scaleFactor, System.Double width, System.Double height, System.Boolean forceExactSize) [0x00029] in <4e388ed9219c471b9cbcf688a60afe69>:0 
  at Xwt.GtkBackend.ImageHandler.ConvertToBitmap (System.Object handle, System.Double width, System.Double height, System.Double scaleFactor, Xwt.Drawing.ImageFormat format) [0x0000c] in <4e388ed9219c471b9cbcf688a60afe69>:0 
  at Xwt.Drawing.Image.ToBitmap (System.Double scaleFactor, Xwt.Drawing.ImageFormat format) [0x00026] in <18f95a9d2a534390b4c0179972e33dcd>:0 
  at Xwt.Drawing.Image.ToBitmap (Xwt.Drawing.ImageFormat format) [0x00000] in <18f95a9d2a534390b4c0179972e33dcd>:0 
  at Xwt.Drawing.ImageBuilder.ToBitmap (Xwt.Drawing.ImageFormat format) [0x00006] in <18f95a9d2a534390b4c0179972e33dcd>:0 
  at Antmicro.Renode.Extensions.Analyzers.Video.FrameBufferDisplayWidget.SetDisplayParameters (System.Int32 desiredWidth, System.Int32 desiredHeight, Antmicro.Renode.Backends.Display.PixelFormat colorFormat, ELFSharp.ELF.Endianess endianess) [0x0006c] in <4650942841e04c76931da7700721cda6>:0 
  at (wrapper remoting-invoke-with-check) Antmicro.Renode.Extensions.Analyzers.Video.FrameBufferDisplayWidget.SetDisplayParameters(int,int,Antmicro.Renode.Backends.Display.PixelFormat,ELFSharp.ELF.Endianess)
  at Antmicro.Renode.Extensions.Analyzers.Video.VideoAnalyzer+<>c__DisplayClass6_1.<OnAttach>b__3 () [0x00000] in <4650942841e04c76931da7700721cda6>:0 
  at Antmicro.Renode.UI.ApplicationExtensions.InvokeInUIThread (System.Action action) [0x00016] in <4650942841e04c76931da7700721cda6>:0 
  at Antmicro.Renode.Extensions.Analyzers.Video.VideoAnalyzer+<>c__DisplayClass6_0.<OnAttach>b__0 (System.Int32 w, System.Int32 h, Antmicro.Renode.Backends.Display.PixelFormat f, ELFSharp.ELF.Endianess e) [0x00005] in <4650942841e04c76931da7700721cda6>:0 
  at Antmicro.Renode.Peripherals.Video.AutoRepaintingVideo.add_ConfigurationChanged (System.Action`4[T1,T2,T3,T4] value) [0x00041] in <30fc4c01ab8a440b89479089b95d632c>:0 
  at Antmicro.Renode.Extensions.Analyzers.Video.VideoAnalyzer.OnAttach (Antmicro.Renode.Backends.Video.VideoBackend backend) [0x0003d] in <4650942841e04c76931da7700721cda6>:0 
  at Antmicro.Renode.UI.GUIPeripheralBackendAnalyzer`1+<>c__DisplayClass2_0[T].<AttachTo>b__0 () [0x00000] in <4650942841e04c76931da7700721cda6>:0 
  at Antmicro.Renode.UI.ApplicationExtensions+<>c__DisplayClass4_0.<InvokeInUIThreadAndWait>b__0 () [0x00000] in <4650942841e04c76931da7700721cda6>:0 
  at Xwt.Application+<>c__DisplayClass19_0.<Invoke>b__0 () [0x0000a] in <18f95a9d2a534390b4c0179972e33dcd>:0
```

swetland, 2022-05-04:
Here's a suitable binary -- it just prints to the serial port and spins forever.  Start it and then poke the framebuffer registers from the monitor as described and boom.

Also, in case it's useful:
This is on an Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz, 32GB ram, GTX1070 GPU (8GB), running Ubuntu 20.04.4 LTS and I'm using the 1.13.0 release of Renode.

[bios.zip](https://github.com/renode/renode/files/8618304/bios.zip)

(obnoxious that github will accept zip files but not tgz...)

swetland, 2022-05-04:
I also verified that the same error occurs if the program writes to the vfb registers instead of doing it from the monitor.

mateusz-holenko, 2022-05-04:
Hi @swetland!

The current model of `LiteX_Framebuffer.cs` supports the 8-bit CSR LiteX configuration in which multi-byte registers are split into 32-bit sub-registers with only lowest 8-bit of meaningful data. This means that in order to configure the `hres/vres` registers you need to do two writes:

```
litex_video WriteDoubleWord 0x1c 0x02
litex_video WriteDoubleWord 0x20 0x80 

litex_video WriteDoubleWord 0x3c 0x01
litex_video WriteDoubleWord 0x40 0xe0
```
What configuration of LiteX are you using?

mateusz-holenko, 2022-05-04:
And thanks for reporting the issue - we'll need to harden this layer to avoid crashing Renode and reporting a meaningful error instead.

swetland, 2022-05-04:
I was not aware such a configuration existed (Litex as I've used it on ULX3S accepts 32bit writes to registers without an issue).  Is this a configuration option somewhere in Renode or does Renode (or the emulated framebuffer) only support this multiple write mode?

mateusz-holenko, 2022-05-04:
This is a configuration of LiteX itself (it's called `--csr-data-width`: https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L328).

Back in the days, when Renode models were created, the 8-bit CSR width configuration was the default one and most of LiteX models support only this setup.
For some selected ones we've already created an alternative implementation targeting the 32-bit CSRs (e.g.: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Timers/LiteX_Timer_CSR32.cs) and we are working on more.

`LiteX_Framebuffer.cs` at the moment works only for 8-bit CSRs - we'll take a look at it. 
BTW, I think you'll also need 32-bit CSR support for other LiteX peripherals (like `LiteX_CPUTimer.cs`).

swetland, 2022-05-04:
Yeah, ideally Renode's peripherals would match the behaviour of the FPGA peripherals.  UART, Timer, and Framebuffer are the most immediately useful.  I'll have to see how involved building Renode from source is -- given an example like two versions of Timer, I can probably adjust things as I trip over them.

I need to double-check but I believe the Framebuffer register layout in Renode doesn't match that in the latest release of Litex.

I suspect one downside of Litex's model of generating headers/docs from the HDL source is they don't worry much about rearranging the hardware.  Maintaining some level of forward/backward compatibility to support software, emulators, etc out-of-tree would be nice...

mithro, 2022-05-07:
@mateusz-holenko - Where can I find the state of CSR registers size support for the current LiteX peripherals?

mithro, 2022-05-07:
Could that info be added to https://docs.google.com/spreadsheets/d/1XTHfdYXuvwoYdPXm4M6qDA0D2fZCPy220-9q6qZpTw4/edit?usp=drivesdk ?

mateusz-holenko, 2022-05-09:
@mithro All LiteX peripherals (i.e., those which names start with `LiteX_`) without the `_CSR32.cs` suffix are the one supporting the 8-bit CSR width configuration. The ones with the suffix support the 32-bit CSR data width.

I'll update the spreadsheet and add links to both versions where applicable.

mateusz-holenko, 2022-05-09:
@swetland We'll verify the layout of registers for LiteX peripherals models and check if they need upgrading.

mateusz-holenko, 2022-06-13:
@swetland, in dc7e33b1b956e03fd4170dd586533f9b05b4b588 we added a model of the LiteX frame buffer supporting the 32-bit CSR width configuration.

We are now working on updating the script for automatic generation of repl/resc files in LiteX.