renode/renode #325: Simulation time and

Regiksy, 2022-05-02:
Hello,
im currently writing a thesis on simulation with Renode and i was hoping to get some 
information regarding Virtual time and Elapsed host time.

I'm running a program on the stm32f103 CPU that toggles an LED  every 200ms.
After i run the simulation for a while i get the following values:
[image]

The program seems to be executing delay correctly.
[image]

The "PerformanceInMIPS" has been set to 1, which according to the documentation should 
cause virtual time to elapse at the same rate as real-world time.
However the Elapesd Vitrual Time and Elapsed Host time differ, what could be the cause of it?
Also do you know what sets the "limit"-value of machine´s Clock source, and does it affect the overall performance of the simulation?

Im running  Renode on Debian 11, inside a VMBox.

snooka2197, 2022-05-04:
Having the same issue.
Also i was expecting the code to be executed 2 times faster when i set the performance = 2. But it still executed at the same speed, although virtual time passes faster. I am checking this with a timer set to 1000ms. What am i missing here?

Xenoamor, 2022-05-26:
Likely the simulation is running slower than the system time so can't keep up. You can change the time steps to allow for faster simulation at the cost of accuracy.

Try running: `emulation SetGlobalQuantum "0.10000"` to set it to be really coarse to see if that changes anything for you