renode/renode #330: SetHookAtBlockEnd breaks emulation

rchtsang, 2022-05-07:
OS: Ubuntu 20.04 VM on OSX Mojave
Renode Version: 1.12.0.33697 (0ced0bb1-202203311843)

Emulated Platform: nrf52840

Files and scripts to recreate the issue: 
https://drive.google.com/drive/folders/1LPUco4lQCB1o0olNXv6DuMXA40WxvXe0?usp=sharing

Issue: 
I am trying to write some custom scripts for getting code coverage but am finding that using SetHookAtBlockEnd causes the emulation to break for some reason. I am testing with a blinky example from nordic's nrf52 SDK v17 with a simply python hook that writes the block's pc to a log file.

This is the output when the python hook is added:
```
(monitor) using sysbus
(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/cpus/nrf52840.repl 
(machine-0) $bin=@/vagrant/examples/nrf52840/blinky/pca10056/blank/armgcc/_build/nr
f52840_xxaa.bin
(machine-0) set logpc """
> with open("/vagrant/logs/blinky.log", 'a') as lf:
>     lf.write("block@{}\n".format(hex(pc)))
> """
(machine-0) cpu SetHookAtBlockEnd $logpc
(machine-0) sysbus LogPeripheralAccess gpio0
(machine-0) macro reset """
> sysbus LoadBinary $bin 0
> """
(machine-0) runMacro $reset
(machine-0) start
Starting emulation...
(machine-0) pause
Pausing emulation...
(machine-0) 
```

```
04:00:50.0743 [INFO] Loaded monitor commands from: /vagrant/renode/scripts/monitor.py
04:01:40.7769 [INFO] System bus created.
04:01:53.5370 [INFO] Reading cache
04:01:53.8556 [INFO] sysbus: Loaded SVD: /tmp/renode-1569/15a5ad44-e181-44ac-b4ec-25e6538910ad.tmp. Name: nrf52840. Description: nRF52840 reference description for radio MCU with ARM 32-bit Cortex-M4 Microcontroller .
04:05:20.2198 [INFO] sysbus: Logging already enabled: False.
04:06:08.1839 [INFO] cpu: Setting initial values: PC = 0x2BD, SP = 0x20040000.
04:06:08.1870 [INFO] machine-0: Machine started.
04:06:08.3741 [WARNING] sysbus: [cpu: 0x3BC] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x0.
04:06:08.3831 [WARNING] sysbus: [cpu: 0x3E0] (tag: 'PSELRESET') ReadDoubleWord from non existing peripheral at 0x10001200, returning 0x0.
04:06:08.3865 [WARNING] sysbus: [cpu: 0x3EE] ReadDoubleWord from non existing peripheral at 0x10001204.
04:06:08.6869 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x234 (PinConfigure+0x34), value 0x3.
04:06:08.6972 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x238 (PinConfigure+0x38), value 0x3.
04:06:08.6978 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x23C (PinConfigure+0x3c), value 0x3.
04:06:08.6980 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x240 (PinConfigure+0x40), value 0x3.
04:06:08.6997 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x8 (OutSet), value 0x2000.
04:06:08.7003 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x8 (OutSet), value 0x4000.
04:06:08.7010 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x8 (OutSet), value 0x8000.
04:06:08.7016 [INFO] gpio0: [cpu: 0x352] WriteUInt32 to 0x8 (OutSet), value 0x10000.
04:06:08.7100 [INFO] gpio0: [cpu: 0x2F8] ReadUInt32 from 0x4 (Out), returned 0x1E000.
04:06:08.7106 [INFO] gpio0: [cpu: 0x2F8] WriteUInt32 to 0x8 (OutSet), value 0x0.
04:06:08.7113 [INFO] gpio0: [cpu: 0x2F8] WriteUInt32 to 0xC (OutClear), value 0x2000.
04:06:11.7198 [INFO] machine-0: Machine paused.
```

Looking at the actual pc values recorded:
```
...
block@0x32cL
block@0x352L
block@0x332L
block@0x34cL
block@0x396L
block@0x2f8L
block@0x3a2L
[... 121 lines cut ...]
03:56:48.0200 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x0.
03:56:48.3232 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0x2000.
03:56:48.3244 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x4000.
03:56:48.3251 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x0.
03:56:48.6201 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0x6000.
03:56:48.6208 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x8000.
03:56:48.6213 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x0.
03:56:48.9311 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0xE000.
03:56:48.9318 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x10000.
03:56:48.9322 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x0.
03:56:49.2118 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0x1E000.
03:56:49.2119 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x0.
03:56:49.2124 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x2000.
03:56:49.5052 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0x1C000.
03:56:49.5058 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x0.
03:56:49.5063 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x4000.
03:56:49.8259 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0x18000.
03:56:49.8267 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x0.
03:56:49.8276 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x8000.
03:56:50.1475 [INFO] gpio0: [cpu: 0x3AC] ReadUInt32 from 0x4 (Out), returned 0x10000.
03:56:50.1478 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0x8 (OutSet), value 0x0.
03:56:50.1480 [INFO] gpio0: [cpu: 0x3AC] WriteUInt32 to 0xC (OutClear), value 0x10000.
03:56:50.2129 [INFO] machine-0: Machine paused.
```
Which evidently does not get stuck.

rchtsang, 2022-05-13:
I have also tried `cpu EnableExecutionTracing` and seem to be met with the same problem.

rchtsang, 2022-05-13:
I'm dumb. I should've paid more attention to the disassembly. 

The `delay_machine_code.0` translates to 
```
0x3800 + NRFX_COREDEP_DELAY_US_LOOP_CYCLES, // SUBS r0, #loop_cycles
0xd8fd, // BHI .-2
0x4770  // BX LR
```
And I remembered that this is what's being used for the `nrf_delay_ms` function, which had been given 500ms. The overhead introduced by logging the PC trace caused the emulator to run a lot slower than without, and I just wasn't giving it enough time to get through the delays. 

I am stupid. Nothing is broken but my brain.

closed 2022-05-13 by rchtsang

PiotrZierhoffer, 2022-05-14:
Thanks for a thorough analysis.

For an approximate result that's visibly faster, try `cpu LogTranslationBlockFetch true` - this only logs when we translate a block, not when we execute it. And we have a translation cache, so blocks are only translated when they're not found cached.

rchtsang, 2022-05-14:
Thanks for the tip!