renode/renode #338: Add extra columns for hps status method in renode

EuniceMadya, 2022-05-24:
**Platform Used:** Debian GNU/Linux

**Renode Version:** 1.13.0.2537

**Platform to run on:** Debian GNU/Linux

Right now, inside [HPSHostController](https://github.com/renode/renode-infrastructure/blob/e6c619cd001c7629d4987021c60e315705ee6e41/src/Emulator/Extensions/Mocks/HPSHostController.cs#L190) only 16 registers' values are read and rendered, while when running `hps status`, we have 19 registers with different values, and therefore we could add few columns for the remaining registers.

PiotrZierhoffer, 2022-05-24:
Sure, we'll look into this!

EuniceMadya, 2022-05-25:
I'm working on it right now, and communicating with the HPS team regarding this too, but any help would be appreciated!

PiotrZierhoffer, 2022-07-13:
@EuniceMadya for the record, are these changes what you were looking for in the context of this issue?

EuniceMadya, 2022-07-20:
Yup looks good!

mateusz-holenko, 2022-07-21:
Great, we'll merge it upstream soon.

mateusz-holenko, 2022-07-22:
Update to the `HPSHostController` is now merged upstream: d6e1ff54acf29d3f22460d181564c43a018569b1.

closed 2022-07-22 by mateusz-holenko