renode/renode #339: Expected behavior

wohlbier, 2022-05-27:
Hi, see below for output when trying to use `platforms/boards/eos-s3-qomu.repl`. Does this look like it is functioning correctly? I am not sure what to make of the `WARNING`s.

Thanks!

```
13:12:23.4744 [INFO] cpu: Entering function memset at 0x4594
13:12:23.4744 [INFO] cpu: Entering function memset at 0x459A
13:12:23.4745 [INFO] cpu: Entering function memset at 0x4594
13:12:23.4745 [INFO] cpu: Entering function memset at 0x4598
13:12:23.4745 [INFO] cpu: Entering function _mainCRTStartup+0x1E (guessed) at 0x1D6
13:12:23.4745 [INFO] cpu: Entering function _mainCRTStartup+0x26 (guessed) at 0x1DE
13:12:23.4746 [INFO] cpu: Entering function _mainCRTStartup+0x2E (guessed) at 0x1E6
13:12:23.4746 [INFO] cpu: Entering function _mainCRTStartup+0x42 (guessed) at 0x1FA
13:12:23.4746 [INFO] cpu: Entering function __libc_init_array (entry) at 0x452C
13:12:23.4747 [INFO] cpu: Entering function __libc_init_array at 0x453C
13:12:23.4747 [INFO] cpu: Entering function _init (entry) at 0x5A08
13:12:23.4747 [INFO] cpu: Entering function __libc_init_array at 0x4544
13:12:23.4748 [INFO] cpu: Entering function __libc_init_array at 0x455A
13:12:23.4749 [INFO] cpu: Entering function frame_dummy (entry) at 0x18C
13:12:23.4750 [INFO] cpu: Entering function frame_dummy+0xE (guessed) at 0x19A
13:12:23.4751 [INFO] cpu: Entering function register_tm_clones (entry) at 0x140
13:12:23.4757 [INFO] cpu: Entering function register_tm_clones+0x16 (guessed) at 0x156
13:12:23.4759 [INFO] cpu: Entering function __libc_init_array at 0x4560
13:12:23.4759 [INFO] cpu: Entering function __libc_init_array at 0x454A
13:12:23.4760 [INFO] cpu: Entering function __libc_init_array at 0x454E
13:12:23.4760 [INFO] cpu: Entering function _mainCRTStartup+0x46 (guessed) at 0x1FE
13:12:23.4760 [INFO] cpu: Entering function main (entry) at 0x8B4
13:12:23.4761 [INFO] cpu: Entering function qomu_hardwaresetup (entry) at 0x42F4
13:12:23.4761 [WARNING] nvic: Unhandled read from offset 0x8.
13:12:23.4762 [INFO] nvic: [cpu: 0x42F4] ReadUInt32 from 0x8 (InterruptControllerType+0x4), returned 0x0.
13:12:23.4782 [INFO] nvic: [cpu: 0x42F4] WriteUInt32 to 0x8 (InterruptControllerType+0x4), value 0x2.
13:12:23.4785 [WARNING] nvic: Unhandled write to offset 0x8, value 0x2.
13:12:23.4786 [INFO] cpu: Entering function S3x_pwrcfg_init (entry) at 0x8D4
13:12:23.4786 [INFO] cpu: Entering function DFS_Initialize (entry) at 0x3AF0
13:12:23.4787 [INFO] cpu: Entering function S3x_pwrcfg_init at 0x8E6
13:12:23.4787 [INFO] cpu: Entering function S3x_Clk_Init (entry) at 0x3650
13:12:23.4787 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc (entry) at 0x2DCC
13:12:23.4787 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc at 0x2DD2
13:12:23.4788 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc at 0x2DE0
13:12:23.4788 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc at 0x2DE8
13:12:23.4788 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc at 0x2DF2
13:12:23.4789 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc at 0x2DF8
13:12:23.4789 [WARNING] sysbus: [cpu: 0x2DF8] (tag: 'AnalogIP') ReadDoubleWord from non existing peripheral at 0x40005480, returning 0x0.
13:12:23.4789 [WARNING] sysbus: [cpu: 0x2DF8] (tag: 'AnalogIP') WriteDoubleWord to non existing peripheral at 0x40005480, value 0x1.
13:12:23.4790 [WARNING] sysbus: [cpu: 0x2DF8] (tag: 'AnalogIP') ReadDoubleWord from non existing peripheral at 0x40005480, returning 0x0.
13:12:23.4791 [WARNING] sysbus: [cpu: 0x2DF8] (tag: 'AnalogIP') WriteDoubleWord to non existing peripheral at 0x40005480, value 0x0.
13:12:23.4792 [WARNING] sysbus: [cpu: 0x2DF8] (tag: 'AnalogIP') WriteDoubleWord to non existing peripheral at 0x40005484, value 0x5B5.
13:12:23.4793 [INFO] cpu: Entering function S3x_Clkd_Change_Hsosc at 0x2E2C
13:12:23.4793 [INFO] cpu: Entering function S3x_Clk_Init at 0x3658
13:12:23.4794 [INFO] cpu: Entering function S3x_Clk_Init at 0x36B0
13:12:23.4795 [INFO] cpu: Entering function S3x_Clk_Init at 0x36B8
13:12:23.4795 [INFO] cpu: Entering function xQueueGenericCreate (entry) at 0x15D0
13:12:23.4796 [INFO] cpu: Entering function xQueueGenericCreate at 0x15E4
13:12:23.4797 [INFO] cpu: Entering function xQueueGenericCreate at 0x162A
13:12:23.4798 [INFO] cpu: Entering function xQueueGenericCreate at 0x15EA
13:12:23.4798 [INFO] cpu: Entering function pvPortMalloc (entry) at 0x28B0
13:12:23.4799 [INFO] cpu: Entering function vTaskSuspendAll (entry) at 0x1CB0
13:12:23.4800 [INFO] cpu: Entering function pvPortMalloc at 0x28C2
13:12:23.4801 [INFO] cpu: Entering function pvPortMalloc at 0x28FA
13:12:23.4801 [INFO] cpu: Entering function prvHeapInit (entry) at 0x27AC
13:12:23.4802 [INFO] cpu: Entering function prvHeapInit at 0x2802
13:12:23.4803 [INFO] cpu: Entering function prvHeapInit at 0x27C2
13:12:23.4803 [INFO] cpu: Entering function prvHeapInit at 0x2800
13:12:23.4804 [INFO] cpu: Entering function pvPortMalloc at 0x28FE
13:12:23.4805 [INFO] cpu: Entering function pvPortMalloc at 0x28C8
13:12:23.4806 [INFO] cpu: Entering function pvPortMalloc at 0x28D0
13:12:23.4806 [INFO] cpu: Entering function pvPortMalloc at 0x28D2
13:12:23.4807 [INFO] cpu: Entering function pvPortMalloc at 0x28DA
13:12:23.4808 [INFO] cpu: Entering function pvPortMalloc at 0x28E4
13:12:23.4808 [INFO] cpu: Entering function pvPortMalloc at 0x28EC
[... 163 lines cut ...]
13:12:23.5021 [INFO] cpu: Entering function __ssputs_r (entry) at 0x4EE0
13:12:23.5094 [INFO] cpu: Entering function __ssputs_r at 0x4F4E
13:12:23.5095 [INFO] cpu: Entering function memmove (entry) at 0x5900
13:12:23.5096 [INFO] cpu: Entering function memmove at 0x5910
13:12:23.5097 [INFO] cpu: Entering function memmove at 0x5928
13:12:23.5097 [INFO] cpu: Entering function __ssputs_r at 0x4F5A
13:12:23.5098 [INFO] cpu: Entering function _printf_common at 0x51F2
13:12:23.5099 [INFO] cpu: Entering function _printf_common at 0x51F6
13:12:23.5100 [INFO] cpu: Entering function _printf_common at 0x5222
13:12:23.5101 [INFO] cpu: Entering function _printf_common at 0x5238
13:12:23.5102 [INFO] cpu: Entering function _printf_i at 0x53C8
13:12:23.5102 [INFO] cpu: Entering function _printf_i at 0x5462
13:12:23.5103 [INFO] cpu: Entering function __ssputs_r (entry) at 0x4EE0
13:12:23.5104 [INFO] cpu: Entering function __ssputs_r at 0x4F4E
13:12:23.5105 [INFO] cpu: Entering function memmove (entry) at 0x5900
13:12:23.5106 [INFO] cpu: Entering function memmove at 0x5910
13:12:23.5106 [INFO] cpu: Entering function memmove at 0x5916
13:12:23.5107 [INFO] cpu: Entering function memmove at 0x5912
13:12:23.5108 [INFO] cpu: Entering function memmove at 0x5928
13:12:23.5109 [INFO] cpu: Entering function __ssputs_r at 0x4F5A
13:12:23.5110 [INFO] cpu: Entering function _printf_i at 0x546C
13:12:23.5110 [INFO] cpu: Entering function _printf_i at 0x5470
13:12:23.5111 [INFO] cpu: Entering function _printf_i at 0x5476
13:12:23.5112 [INFO] cpu: Entering function _printf_i a
```

mateusz-holenko, 2022-05-31:
Hi @wohlbier, in Renode warnings about unhandled accesses to peripherals do not necessary mean serious problems. This is caused by the fact that platforms are often implemented partially as not all elements are required to correctly simulate a given piece of software.

In this example I see mostly warnings about accesses to the clock control system - this is a layer we typically don't need to implement.

That being said it might happen that when you change a simulated binary you might be required to extend the model logic and/or platform definition.

The question here is if you see your software not behave in the simulation as expected?

wohlbier, 2022-05-31:
@mateusz-holenko thanks for the response! I am in fact not sure whether the simulation is operating correctly. The main program does very little at the moment, and I am trying to incrementally add functionality and understand the results. The main program I'm building off can be seen here: https://github.com/QuickLogic-Corp/qorc-sdk/blob/master/qomu_apps/qomu_helloworldsw/src/main.c
At the moment I'm calling qomu_hardwaresetup and nvic_init. I don't really know what to look for to see if these calls are successful. Any tips are appreciated! Thanks!

PiotrZierhoffer, 2022-05-31:
While Renode offers tons of different debugging features (take a look at docs.renode.io, esp the section about GDB), you can see some interesting output with `cpu LogFunctionNames true true` (the latter `true` removes duplicate entries).

This will give you some insight into where's your application going. 

Also `sysbus LogAllPeripheralsAccesses true` will give you information about all reads and writes to memory mapped devices.