renode/renode #340: How to connect simulated UART peripheral

Xenoamor, 2022-05-27:
How can I attach a uart peripheral written in a .cs file to a machine in its repl file such as `gps: UART.gps @ uart1`?
There's methods of doing this with both I2C and SPI but I can not find one for UART.

I'm aware I could spin up an entirely new machine and connect it through a UART hub but this seems overkill as I'd have to write code for an Arm Cortex and then emulate that. I've also tried using TCP and terminal access which does work but has no timing synchronisation options.

I've tried the following but it fails to connect with the error "Register 'uart1' of type 'Antmicro.Renode.Peripherals.UART.NRF52840_UART' does not provide an interface to register 'uart_loopback'
```
using System;
using System.Collections.Generic;
using Antmicro.Renode.Core;
using Antmicro.Renode.Core.Structure.Registers;
using Antmicro.Renode.Logging;
using Antmicro.Renode.Utilities;
using Antmicro.Renode.Peripherals.Bus;
using Antmicro.Migrant;

namespace Antmicro.Renode.Peripherals.UART
{
    //This is a dummy device, emulating connection between TX and RX lines.
    public class UARTLoopback : IUART
    {
        public UARTLoopback()
        {
            this.CharReceived += WriteChar;
        }

        public void Reset()
        {
        }

        [field: Transient]
        public event Action<byte> CharReceived;
    
        public void WriteChar(byte value)
        {
        }

        public Bits StopBits => Bits.One;

        public Parity ParityBit
        {
            get
            {
                return Parity.None;
            }
        }

        public uint BaudRate => 0;
    }
}
```

Is there some other method I'm missing here?

PiotrZierhoffer, 2022-05-31:
I assume that what you want to achieve is to have a GPS device connected over uart to the simulated platform.

Usually we treat UART devices as serial output or a mean to communicate between machines - your analysis is correct. UARTHub would help, but it's not perfect, as you still need to register the peripheral somewhere.

A solution that would probably fit your need is implemented in https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/UART/EFM32_UART.cs

Specifically, you are interested in having the `IPeripheralContainer<IUART, NullRegistrationPoint>` interface implemented, with the following methods:

```
        private IUART registeredPeripheral;

        public void Register(IUART peripheral, NullRegistrationPoint registrationPoint)
        {
            if(registeredPeripheral != null)
            {
                throw new RegistrationException("Cannot register more than one peripheral.");
            }
            Machine.RegisterAsAChildOf(this, peripheral, registrationPoint);
            registeredPeripheral = peripheral;
        }

        public void Unregister(IUART peripheral)
        {
            Machine.UnregisterAsAChildOf(this, peripheral);
            registeredPeripheral = null;
        }

        public IEnumerable<NullRegistrationPoint> GetRegistrationPoints(IUART peripheral)
        {
            return new [] { NullRegistrationPoint.Instance };
        }

        public IEnumerable<IRegistered<IUART, NullRegistrationPoint>> Children
        {
            get
            {
                return new []  { Registered.Create(registeredPeripheral, NullRegistrationPoint.Instance) };
            }
        }
```

That implementation is not complete, the `Register` method would also need to connect the events, something like:

```
this.CharReceived += peripheral.WriteChar;
peripheral.CharReceived += this.WriteChar;
```
and you would most likely be interested in adding some locking.

Then you should be able to do `gps: UART.gps @ uart1`.

I have not tried it in code, but it would be my best guest to move forward.

If you find this solution to be helpful, then it's something we might consider adding on a generic layer (in UARTBase).

jonlamb-gh, 2023-10-09:
FWIW, `UARTBase` appears to have all that's needed now.

```
namespace Antmicro.Renode.Peripherals.UART
{
    public class PMS5003 : UARTBase
    ...
```

JohnZhang3IoT, 2024-12-18:
> FWIW, `UARTBase` appears to have all that's needed now.
> 
> ```
> namespace Antmicro.Renode.Peripherals.UART
> {
>     public class PMS5003 : UARTBase
>     ...
> ```

Hi @jonlamb-gh , may I ask after that, how to register the device in the resc file? I tried xxx:XXX @uart1, but it returns the error:
Error E20: Undefined register '[ReferenceValue: uart1]'.

jonlamb-gh, 2024-12-18:
> Hi @jonlamb-gh , may I ask after that, how to register the device in the resc file? I tried xxx:XXX @uart1, but it returns the error: Error E20: Undefined register '[ReferenceValue: uart1]'.

Hey @JohnZhang3IoT, I'm not sure what exactly that error means in the context of your setup. In case it helps, here's where I'm using that locally-defined peripheral: https://github.com/jonlamb-gh/air-gradient-pro-rs/blob/a9cc261fda046c771c533aaaf94ac6707cf9465d/renode/agp.resc#L52-L53

JohnZhang3IoT, 2024-12-19:
> Hey @JohnZhang3IoT, I'm not sure what exactly that error means in the context of your setup. In case it helps, here's where I'm using that locally-defined peripheral: [...]

Hi @jonlamb-gh,  Great thanks for your information, it helps a lot, I should use @usart1, then error become 
```
There was an error executing command 'machine LoadPlatformDescriptionFromString "PMS5003:UART.PMS5003 @ usart1"'
Error E05: Register 'usart1' of type 'Antmicro.Renode.Peripherals.UART.STM32_UART' does not 
provide an interface to register 'PMS5003' of type 'Antmicro.Renode.Peripherals.UART.PMS5003'.
At 1:24:PMS5003:UART.PMS5003 @ usart1
                                                         ^^^^^^
```
maybe in your case work and my case not work is related to STM32_UART, I will study your STM32_UART_CUSTOM code, thanks again!