renode/renode #342: Example of AddUserStateHook "state"

gonbbb, 2022-06-08:
Hello, 

I would like to use a hook called AddUserStateHook, but I don't have a concrete example of a state. Is there a list somewhere?
I think there are several types of `states`, but I don't know if there is a list somewhere.

```
(machine) machine AddUserStateHook "state" "print 'User state has changed to: %s state '"
```

Examples of state could be `running`, `stop`, `halt`, etc., but I am not sure if they are correct.

> User state hooks enable you to code a reaction to the event when “UserState” string of a machine changes.

Or, as above, are there any other monitor scripts that determine CPU status?

ref. https://renode.readthedocs.io/en/latest/basic/using-python.html#user-state-hooks

PiotrZierhoffer, 2022-06-14:
Hi @gonbbb 

Unfortunately, there are no examples of user state hooks - we use it on some custom occasions, that are difficult to generalize and describe properly.

The "state" is an arbitrary string - it can be anything you want.

A possible use case is when you have a series of other hooks that verify some conditions and set the user state, then have a separate python script as user state hook, that will analyze the current state and react accordingly. So in a way this can be used as a hacky means of communication between hook scripts.

gonbbb, 2022-06-23:
@PiotrZierhoffer 
Sorry for the delay in replying.

I was testing Renode's hook feature. 

I thought that state here is similar to the following permark.
https://github.com/antmicro/renode-riscv-torture-example/blob/6b8c8d5fb2626d6fb4af3b64be159086ad26ca15/torture.resc#L31