renode/renode #344: Pass a value to an RESC script

barneywilliams, 2022-06-16:
We have a setup that allows us to execute test suites written in a C/C++ test framework. Below is the .resc script:
```
mach create
machine LoadPlatformDescription @platforms/cpus/nxp-k6xf.repl

macro reset
"""
    sysbus LoadELF @${TESTSUITE}
"""

runMacro $reset

start

uart_connect sysbus.lpuart1
```

Is there any way we could pass a filename in as a command-line argument or an environment variable? Maybe some special syntax via Python?

Such a feature would be very helpful for other parameters as well. I have not been able to find such a mechanism, if it does exist.  Thanks in advance for your help!

PiotrZierhoffer, 2022-06-16:
Hi @barneywilliams 

If you want to pass a variable to Renode, the easiest way would be like this:

script.resc:
```
mach create
machine LoadPlatformDescription @platforms/cpus/nxp-k6xf.repl

macro reset
"""
    sysbus LoadELF $bin
"""

runMacro $reset

start

uart_connect sysbus.lpuart1
```

This is basically your script with a change in LoadELF line

Then run Renode with 

`renode -e '$bin=@path/to/your.elf; include @script.resc'`

Would that suffice in your use case?

barneywilliams, 2022-06-17:
That is perfect. Thanks!

closed 2022-06-17 by barneywilliams