renode/renode #345: Compiling on windows fails

mcine, 2022-06-17:
I am trying to compile renode in windows and I am following this guide: https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html

I have installed cygwin and added few components that were not mentioned in this guide and also winpthread (all packages containing that with sources) that seems to be the problem at the moment. I installed cygwin with online installer.

build.sh process seemsto be going ok untill this:
C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): warning : /usr/lib/gcc/x86_64-pc-cygwin/11/../../../../x86_6
4-pc-cygwin/bin/ld: cannot find -lwinpthread: No such file or directory
C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): error : collect2: error: ld returned 1 exit status

Complete list of warnings and errors from build.sh are at the end of this message.

I tried also the solution file with VS 2022 (64bit). It says that there is errors in configuration mappings but I did not notice anything od in the configuration manager..
when building, I get bunch of missing files:
34>------ Rebuild All started: Project: MonitorTests, Configuration: Debug Any CPU ------
34>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\lib\AntShell\AntShell\bin\Debug\AntShell.dll' could not be found
34>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\bin\Debug\Emulator.dll' could not be found
34>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\src\Infrastructure\src\Emulator\Extensions\bin\Debug\Extensions.dll' could not be found
34>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\lib\Migrant\Migrant\bin\Debug\Migrant.dll' could not be found
34>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\src\Infrastructure\src\Emulator\Peripherals\bin\Debug\Renode-peripherals.dll' could not be found
34>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\output\bin\Debug\Renode.exe' could not be found
33>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\lib\AntShell\AntShell\bin\Debug\AntShell.dll' could not be found
33>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\bin\Debug\Emulator.dll' could not be found
33>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\src\Infrastructure\src\Emulator\Extensions\bin\Debug\Extensions.dll' could not be found
33>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\output\bin\Debug\Renode.exe' could not be found
33>CSC : error CS0006: Metadata file 'C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Tests\UnitTests\bin\Debug\UnitTests.dll' could not be found
========== Rebuild All: 4 succeeded, 20 failed, 10 skipped ==========

build.sh warnings and errors:
Build FAILED.

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\UI\UI.csproj" (default target) (3:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Emulator.csproj" (default target) (4:3) ->
(CoreCompile target) ->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Utilities\VideoCapturer.cs(242,29): warning CS0169: The field 'VideoCapturer.starte
d' is never used [C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Emulator.csproj]
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Utilities\VideoCapturer.cs(241,28): warning CS0649: Field 'VideoCapturer.fd' is nev
er assigned to, and will always have its default value 0 [C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Emulator.csproj]
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Utilities\VideoCapturer.cs(244,42): warning CS0649: Field 'VideoCapturer.loggingPar
ent' is never assigned to, and will always have its default value null [C:\projects\renode\renode\src\Infrastructure\src\Emulator\Main\Emulator.cspr
oj]

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\UI\UI.csproj" (default target) (3:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Extensions\Extensions.csproj" (default target) (7:3) ->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Extensions\HostInterfaces\Camera\HostCamera.cs(257,33): warning CS0649: Field 'HostCamer
a.device' is never assigned to, and will always have its default value null [C:\projects\renode\renode\src\Infrastructure\src\Emulator\Extensions\Ex
tensions.csproj]

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\cores-arm.csproj" (default target) (11:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj" (VerifyProperties;GenerateFlags;Compile;Link;Build target) (33) ->
(Link target) ->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): warning : /usr/lib/gcc/x86_64-pc-cygwin/11/../../../../x86
_64-pc-cygwin/bin/ld: cannot find -lwinpthread: No such file or directory

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\cores-arm-m.csproj" (default target) (12:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj" (VerifyProperties;GenerateFlags;Compile;Link;Build target) (33:2)
->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): warning : /usr/lib/gcc/x86_64-pc-cygwin/11/../../../../x86
_64-pc-cygwin/bin/ld: cannot find -lwinpthread: No such file or directory

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\cores-i386.csproj" (default target) (13:2) ->
[... 103 lines cut ...]
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): error : collect2: error: ld returned 1 exit status

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\cores-riscv.csproj" (default target) (17:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj" (VerifyProperties;GenerateFlags;Compile;Link;Build target) (33:7)
->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): error : collect2: error: ld returned 1 exit status

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\cores-riscv64.csproj" (default target) (24:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj" (VerifyProperties;GenerateFlags;Compile;Link;Build target) (33:8)
->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): error : collect2: error: ld returned 1 exit status

"C:\projects\renode\renode\Renode.sln" (default target) (1) ->
"C:\projects\renode\renode\src\Renode\Renode.csproj" (default target) (2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\cores-xtensa.csproj" (default target) (27:2) ->
"C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj" (VerifyProperties;GenerateFlags;Compile;Link;Build target) (33:9)
->
  C:\projects\renode\renode\src\Infrastructure\src\Emulator\Cores\translate.cproj(130,7): error : collect2: error: ld returned 1 exit status

    15 Warning(s)
    11 Error(s)

mcine, 2022-06-17:
tried to compile optionsparser manually in visual studio, but it it skipped all the time for some reason, even when I put that as a startup project. .

PiotrZierhoffer, 2022-06-21:
It seems you're not using mingw compiler, but cygwin-native GCC.

Please verify you have mingw in your `$PATH`, the path should be something like
`/cygdrive/c/Program Files/mingw-w64/x86_64-8.1.0-win32-sjlj-rt_v6-rev0/mingw64/bin`

mcine, 2022-06-22:
Ah yes, thanks, got it compiling now.
Also I got the solution to  work. I think that many of the projects were skipped because renode project had x64 for architecture. works with anycpu.
Not 100% sure if this is all that it needs, since I compiled many projects with Rider before finding that out. VS did not give any good message about this issua.

closed 2022-06-22 by mcine