renode/renode #347: How to connect bitbanding bus to a gpio-port

mcine, 2022-06-22:
Hey,

I saw that you had bitbanding functionality included, but I have trouble understanding how to configure it to work. But it is somehow generated, is it only for specific purpose ?
I have a PIO-bus that has sw I2C-bus (data and clock) connected and few peripherals connected to those.
pio bus has own register to set, clear and get bit status. 

If I understood correctly. I need to support my gpio to get correct bytes to some memory address (should be the bit-status, but I cannot use that since it is part of my pio registers) and then somehow connect the bitbanding peripheral to it and it will map the data to another bus address where I can configure the peripherals?

To me it also seems that write operation happens for every bit. What does the peripherals need to use that so that they only get word / byte write operations?

Or have I totally misunderstood this.. would be nice to have even some comments in the code files.

- mcine

mcine, 2022-07-01:
I have been working on this and when I send data to peripheral using Write(byte[]), the peripheral only receives 2 bytes of data?
I print my array length at sender and it is ok, but the write function data length is always 2.

I have seen other i2cperipherals to handle more data than 2 in write functions so I assume that it should work... 

Since I am just including my code to binary renode, I think I have no good way around this, since I think I cannot make common interface, since the imported code does not seem to see each other..

Any ideas how to fix this ?

PiotrZierhoffer, 2022-07-05:
Hi @mcine 

There seem to be quite a few issues at the same time here.

Regarding bitbanding - the granularity of the final access depends on the interfaces you support in the target peripheral: IBytePeripheral, IWordPeripheral or IDoubleWordPeripheral.

We can't actually do bit-accurate accesses at the moment, it's always "read-modify-write" for bitband writes.

Regarding the I2C interface, I'd need more data on the platform you use etc.

PiotrZierhoffer, 2022-07-05:
to use types between files you include you can try:

```
i @first_file
EnsureTypeIsLoaded "Namespace.YourFirstType"
i @second_file_depending_on_the_first
```

But again, I'd need more details

mcine, 2022-08-08:
There is no i2c interface, it is all done with sw.. and I am supporting that . I think I got it somehow working before holidays.. 
I have made  my connection to i2c like this.. 
pioa: GPIOPort.SAM4S_GPIO @ sysbus 0x400E0E00
    IRQ -> nvic@11
    1 -> i2c@0 // clock
    2 -> i2c@1 // data

and I have my own i2c implementation:
i2c: I2C.GPIOI2CBus @ pioa 0
    DATA -> pioa@2

one issue here is that since I am handling i2c connection in onGPIO function, I cannot use gpio interface to change the data line status for ACK (locked by the gpio write function) and I need to keep internal status, but the status is checked by reading so it should not be a problem..

the longer i2c message support problem I solved by checking if the device is bytedevice and in that case write every word separately, which seems to work.. But I still am nos sure, why this does not work: selectedPeripheral?.Write(toPeripheralBuffer.ToArray()); even tho the array has many element and the write takes array of bytes. I only receive2 bytes. selectedperipheral is II2CPeripheral

I tried the EnsureTypeIsLoaded command,but still if I have "if (selectedPeripheral is MyPeripheral)" , I get "given type was not found" error  when loading new script. Fails at Antmicro.Renode.Utilities.TypeManager.GetTypeByName(String name, Func`2 assemblyResolver)