renode/renode #349: MIMXRT1064 GPIO

csatadani, 2022-06-28:
Hi! I am using the NXP MIMXRT1064-EVK with renode. I came across the issue, that the gpios don't change their state. I wrote a basic blinky program in which I used the user button and the green led. But even if I set the GPIO pin of the led directly it still doesn't change. Same with the button: pressing it doesn't change the state of the pin so I constantly read 0 out of it. I might be doing something wrong but I don't know what.
Could you please help me out?

Thank you in advance!

PiotrZierhoffer, 2022-07-05:
@csatadani thanks for the report.

Can you please attach the binary so that we can replicate your issue?

csatadani, 2022-07-06:
Of course I can!
Here is my binary:
Thank you for looking into it!
[temp.tar.gz](https://github.com/renode/renode/files/9053048/temp.tar.gz)
:

csatadani, 2022-07-06:
Also for what i have gathered the problem might be that the Register Map is not implemented specifically for the GPIOs of the NXP MIMXRT1064-EVK board, like in the STM32f4 board.

mateusz-holenko, 2022-09-02:
This should be fixed by https://github.com/renode/renode/commit/b8d1a66028e0a59dc017f5e6555e09e63fefed32, @csatadani could you check if it works fine for you now?

csatadani, 2022-09-12:
Thank you! It works just fine now!