renode/renode #350: zephyr v3.0 doesn't work for MIMXRT1064-evk

GaelRap, 2022-06-29:
I'm trying to run out the samples on

rgraczyk, 2022-08-02:
Hi, I have similar problem with Zephyr 3.1 on renode 1.13.1.
Script the same as in earlier post.

```
17:13:45 [INFO] System bus created.
17:13:45 [INFO] sysbus: Loading segment of 28688 bytes length at 0x70000000.
17:13:45 [INFO] sysbus: Loading segment of 60 bytes length at 0x70007010.
17:13:45 [INFO] sysbus: Loading segment of 4096 bytes length at 0x80000040.
17:13:54 [INFO] cpu: Setting initial values: PC = 0x700033B9, SP = 0x80000640.
17:13:54 [INFO] mxrt1064-eval: Machine started.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD28, value 0xFFFFFFFF.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD2C, value 0xFFFFFFFF.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD94, value 0x0.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD98, value 0x0.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD9C, value 0x70000010.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xDA0, value 0x702002B.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD98, value 0x1.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD9C, value 0x80000011.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xDA0, value 0x110B0031.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xD94, value 0x5.
17:13:54 [WARNING] nvic: Unhandled write to offset 0xF50, value 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002E92] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC018, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002E92] (tag: 'CCM(CCM)') WriteDoubleWord to non existing peripheral at 0x400FC018, value 0x1000.
17:13:54 [WARNING] sysbus: [cpu: 0x70002E92] (tag: 'CCM(CCM)/CCM_CBCDR') ReadDoubleWord from non existing peripheral at 0x400FC014, returning 0xA8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002E92] (tag: 'CCM(CCM)/CCM_CBCDR') WriteDoubleWord to non existing peripheral at 0x400FC014, value 0x20A8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002E92] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC048, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002ED6] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC010, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002ED6] (tag: 'CCM(CCM)') WriteDoubleWord to non existing peripheral at 0x400FC010, value 0x1.
17:13:54 [WARNING] sysbus: [cpu: 0x70002ED6] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC048, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EEA] (tag: 'CCM(CCM)/CCM_CBCDR') ReadDoubleWord from non existing peripheral at 0x400FC014, returning 0xA8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EEA] (tag: 'CCM(CCM)/CCM_CBCDR') WriteDoubleWord to non existing peripheral at 0x400FC014, value 0xA8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EEA] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC048, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)/CCM_CBCDR') ReadDoubleWord from non existing peripheral at 0x400FC014, returning 0xA8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)/CCM_CBCDR') WriteDoubleWord to non existing peripheral at 0x400FC014, value 0xA8300.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC018, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)') WriteDoubleWord to non existing peripheral at 0x400FC018, value 0xC0000.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)/CCM_CBCDR') ReadDoubleWord from non existing peripheral at 0x400FC014, returning 0xA8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)/CCM_CBCDR') WriteDoubleWord to non existing peripheral at 0x400FC014, value 0xA8200.
17:13:54 [WARNING] sysbus: [cpu: 0x70002EFA] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC048, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002F1C] (tag: 'CCM(CCM)/CCM(CSCDR1)') ReadDoubleWord from non existing peripheral at 0x400FC024, returning 0x40.
17:13:54 [WARNING] sysbus: [cpu: 0x70002F1C] (tag: 'CCM(CCM)/CCM(CSCDR1)') WriteDoubleWord to non existing peripheral at 0x400FC024, value 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002F1C] (tag: 'CCM(CCM)/CCM(CSCDR1)') ReadDoubleWord from non existing peripheral at 0x400FC024, returning 0x40.
17:13:54 [WARNING] sysbus: [cpu: 0x70002F1C] (tag: 'CCM(CCM)/CCM(CSCDR1)') WriteDoubleWord to non existing peripheral at 0x400FC024, value 0x40.
17:13:54 [WARNING] sysbus: [cpu: 0x70002F1C] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC054, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70002F1C] (tag: 'CCM(CCM)') WriteDoubleWord to non existing peripheral at 0x400FC054, value 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x700038A8] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC078, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x700038A8] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC078, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x700038A8] (tag: 'CCM(CCM)') WriteDoubleWord to non existing peripheral at 0x400FC078, value 0xC.
17:13:54 [WARNING] sysbus: [cpu: 0x700038C2] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC070, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x700038C2] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC070, returning 0x0.
[... cut ...] LIN Break Detect Interrupt Flag (0x1).
17:13:54 [WARNING] sysbus: [cpu: 0x70003D74] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC06C, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70003D74] (tag: 'CCM(CCM)') ReadDoubleWord from non existing peripheral at 0x400FC06C, returning 0x0.
17:13:54 [WARNING] sysbus: [cpu: 0x70003D74] (tag: 'CCM(CCM)') WriteDoubleWord to non existing peripheral at 0x400FC06C, value 0x300000.
17:13:58 [INFO] mxrt1064-eval: Machine paused.
```

mateusz-holenko, 2024-04-30:
Hi, thanks for reporting the issue.

We've double checked it with the current Zephyr version (3.6) and the current Renode (1.15) and all basic Zephyr samples, including hello world, seem to be simulated correctly now.
You can check the status on https://zephyr-dashboard.renode.io/.

closed 2024-07-12 by PiotrZierhoffer