renode/renode #360: How to receive file via TFTP on the simulation side

tonylitianyu, 2022-07-20:
Hi, I am new to Renode. I would like to do cross-compilation on host and pass the file to the simulation. The Renode documentation provides a great [example](https://renode.readthedocs.io/en/latest/host-integration/sharing-files.html?highlight=transferring%20files#starting-the-tftp-server) of how to set up TFTP on Renode monitor. However, after I use 

```
server.tftp ServeFile @path/to/file "filename"
```

I am not sure what to do after that for my embedded simulation to receive the file and run it.

I understand this might be cpu/board specific. Is there a general setup for the simulation side that needs to be done in order to receive the file? Any advice/hint would be great! Thanks!

PiotrZierhoffer, 2022-07-20:
Hi @tonylitianyu 

In short, you need the following components:
- a file you want to share
- a tftp server `emulation CreateNetworkServer "server" "192.168.100.100"; server StartTFTP 69; server.tftp ServeFile @path/to/file "filename"`. The IP has to be in the same network as your platform.
- a switch `emulation CreateSwitch "switch"`
- a connection between the server and the switch `connector Connect server switch`
- a platform with an Ethernet controller. In this aspect this is board specific, but even if your platform does not support Ethernet, you can still add a controller to the platform definition in Renode. This will, in turn, require your software to enable and use this support, so it's a longer discussion. Let's assume it's there and it's called "sysbus.ethernet" in your platform
- connection between the Ethernet controller and the switch `connector Connect sysbus.ethernet switch`

With this setup, after you start the simulation, your software should be able to configure the Ethernet port and fetch the file via tftp. Of course this part will be specific to your solution.

A side note: what exactly do you want to achieve? If it's just about "having a file from host appear in memory" you can always do `sysbus LoadBinary @path/to/file 0x{address}`.

tonylitianyu, 2022-07-27:
I am trying to use RCC in [another repo of yours](https://github.com/antmicro/renode-rtems-leon3) 
However, after compiling, I am not sure what is a good way to transfer file from the host to renode rtems. Leon3 has an ethernet port, it seems like I can utilize that with TFTP.

I just want to try a hello world example for now.

PiotrZierhoffer, 2022-08-02:
I'm not sure about RTEMS support for TFTP, but there is another way if you want - you can transfer the file via UART. 

Please take a look here: https://renode.readthedocs.io/en/latest/host-integration/uart.html

You can either connect to your host via a pseudoterminal or a socket. This should be way easier than supporting ethernet and TFTP.

PiotrZierhoffer, 2023-01-08:
I'm closing the issue for now - if you have any new input, feel free to comment here or create a new one

closed 2023-01-08 by PiotrZierhoffer