renode/renode #361: nRF52840 Watchdog Timer not working (ignores valid reloads)

rene-becker-setec, 2022-07-25:
We are trying to run some FW that utilizes the nRF52840 HW watchdog. Even though the FW is running fine on an nRF52840 Dev Kit, when running in Renode we are seeing periodic resets due to the watchdog. This is using Renode V1.13

The behaviour can easily be replicated by using Zephyr's Task Watchdog sample compiled for the nRF52DK board. The following screenshot shows what happens if I modify the Zephyr sample App to never 'get stuck':
[image]

The following is showing the modification to the Zephyr sample (sample located at `zephyr/samples/subsys/task_wdt/`, using Zephyr v3.1.0):

[image]

Going through the Watchdog source code I see the following problems:
- There is a function to determine whether the Watchdog counter should be reloaded
  
  [image]

  however, this function is never called upon.
  I believe this function should be called whenever a (successful) write to one of the RR registers occurs, i.e. here:
  
  [image]

- The logic of the Reload function is not correct. I had to modify the logic as follows to get it to work
  
  [image]

  for copy & paste: 

  ```C#
       private void Reload()
        {
            if(readyToReset)
            {
                // readyToReset flag is true, which means that we already timeouted, ignore reload
                this.Log(LogLevel.Warning, "Trying to reload after timeout, ignored");
                return;
            }

            var doReload = false;
            for(var i = 0; i < NumberOfRegisters; i++)
            {
                doReload |= (requestRegisterEnabled[i] && !requestRegisterStatus[i].Value);
            }

            if(doReload)
            {
                for(var i = 0; i < NumberOfRegisters; i++)
                {
                    requestRegisterStatus[i].Value = requestRegisterEnabled[i];
                }
                Value = Limit + 1;
                this.NoisyLog("Counter reloaded");
            }
        }
  ```

With the above 2 modifications I get the simulation to behave exactly as the HW using the stock Zephyr task watchdog sample app:
[image]

rene-becker-setec, 2022-08-02:
Awesome. Thanks

PiotrZierhoffer, 2022-08-14:
@rene-becker-setec you will find the fix on master with ec3efdf

closed 2022-08-14 by PiotrZierhoffer

rene-becker-setec, 2022-08-14:
Great! Thanks!