renode/renode #364: Creating a board visualization

mkschreder, 2022-08-04:
I have a rough idea of how this would be done but need some help. 

What would be the proper steps to add a gui window to renode that is able to access peripheral state and display it in realtime as a graphical UI? 

Can you clarify the relationships between different parts of renode where changes would need to be added to make this work?

PiotrZierhoffer, 2022-08-05:
Hi @mkschreder 

This is a very broad question and depends on your goals. 

For some reference on displaying windows in Renode you can look at https://github.com/renode/renode-infrastructure/tree/45c593d4f17320006a05f4bebe8ce8f860f2263b/src/Plugins/AdvancedLoggerViewerPlugin

A somewhat easier approach, that's probably more useful for visualization, can be found here: https://github.com/antmicro/renode-board-visualization

This repo is an exampe of web-based visualization of Arduino Nano 33, with the page served by a Python HTTP server running in Renode (see https://github.com/antmicro/renode-board-visualization/blob/master/vis.py). Please note it's a proof-of-concept rather than a complete, robust solution.

mkschreder, 2022-08-07:
@PiotrZierhoffer ah very nice. Logger viewer was exactly what I was looking for. I like the python approach as well. 

What would be a good way to get a detailed overview of the python bindings? In particular what is being exported from clr and what is not?