renode/renode #366: Creating Network for many instance of Linux VexRISCV

mrinformatics, 2022-08-09:
I recently started using the renode, and was wondering how can I connect various linux instances running using mach create. Also, from what I understand the rootfs is being downloaded from the server, and on every simulation it prompts 

'Starting network: ip: socket: Function not implemented
ip: socket: Function not implemented
FAIL'

Is there any way to get it working?

Any help will be appreciated. 
Thanks in advance.

MateuszKarlic, 2022-11-14:
Hi @thesaadmemon,

The demo you are running targets older kernel version without the networking support enabled.  We plan on updating it in the future, but for now you could either use a different demo with the kernel that has the support (`scripts/single-node/litex_vexriscv_sdcard.resc`), or build a new kernel and rootfs instead, as described here: https://github.com/renode/renode/issues/374#issuecomment-1254859841.

After building, you can use `litex_json2renode.py` from https://github.com/enjoy-digital/litex to generate a new platform description (repl), which will automatically include an Ethernet adapter. Then you should be able to use the adapter.

PiotrZierhoffer, 2023-01-10:
Closing for now. Feel free to add a comment if you want to discuss this further.

closed 2023-01-10 by PiotrZierhoffer